JAndrassy/ArduinoOTA #1: nRF52 test

jandrassy, 2019-01-01:
nRF52 was not tested yet. it compiles.

sigalabs, 2021-01-23:
i am trying to test it on a nrf52832 board and when it's start crashing and restarting. Tried with Sandeep nrf52 core with and without a SoftDevice.

jandrassy, 2021-01-24:
thank you. please try without soft device with extras/nRF5/platform.local.txt from GitHub. I just updated it with build extra flags for no-delete-null-pointer-checks

sigalabs, 2021-01-24:
Hi @jandrassy , thanks for your response, i can confirm that its working great on the nrf52832 chip. I am trying now with the a softdevice. 
Thank you!

sigalabs, 2021-01-24:
With the softdevice i have some issues, i am trying with the BLESerial example ... before trying OTA i am stopping the service using bleSerial.end(); function then trying to do the OTA and the  MCU is crashing and restarts.
Any thought on this? thank you very much!

jandrassy, 2021-01-24:
thank you. partially good news.

did you build and upload the sketch with softdevice selected in Tools menu?

sigalabs, 2021-01-24:
Yes, i am using softdevice 2.0.1 and BLESerial example... i am sure it has to do with SoftDevice usage, if i dont use the bleSerial.start() it's ok... probably i need somehow to disable completely the Softdevice, i hoped that bleSerial.end();  would do the work but its crashing. I ll keep trying and let you know.

jandrassy, 2021-01-24:
why `bleSerial`? only peripheral mode of BLEPeripheral library is supported.

sigalabs, 2021-01-26:
I am using this one example
https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/master/examples/serial

to implement a virtual Serial between MCU and mobile app. Its working great till now with the nrf52, only the .end() function is not stopping the BLE service for some reason.

jandrassy, 2021-01-26:
try to ask Sandeep in the library repository

sigalabs, 2021-01-27:
I have found a solution, i am using the SD function sd_ble_gap_adv_stop(); just before the .end() and it stops advertising now.
When the BLE is stopped that way your library is working great on the nRF52832 chip, thank you very much for your support.

closed 2021-01-27 by jandrassy