JAndrassy/ArduinoOTA #5: ESP32 OTA update via Ethernet (UIPEthernet)

frederic-bonjour, 2019-03-28:
Hi there,

I'd like to know if it's possible to update a sketch on an ESP32 via Ethernet with a ENC28J60 module? I'm using the UIPEthernet library.

The _OTEthernet_ example does not compile, I got the following error:

> 'InternalStorage'` was not declared in this scope

I tried to `#include <InternalStorage.h>`, and the error changes to:

> no matching function for call to 'ArduinoOTAClass::begin(IPAddress, const char [8], const char [9], InternalStorageClass&)'

Any idea?

Thanks! ;)

jandrassy, 2019-03-28:
esp32 and esp8266 have their own ArduinoOTA library, which uses the core's Updater object. it would be easy to enhance this library for esp8266 and esp32 Arduino core, but it is not done yet

frederic-bonjour, 2019-03-28:
OK, thank you for your time!

jandrassy, 2019-04-14:
ESP32 and ESP8266 support added in 1.0.1

closed 2019-04-14 by jandrassy