JAndrassy/ArduinoOTA #6: Arduino Mega 2560 - OTA failed

peros550, 2019-05-12:
I tried to push a new firmware using the Arduino GUI on a Arduino Mega 2560 board. I had already flashed a code with your example, therefore I found a new network port on the GUI successfully. 

During the OTA flashing a new window came up asking my password. I gave the correct password but then the password window came up again and again like I had given a wrong password. 

It looks that for me the process cannot complete successfully. 

In the Arduino GUI there was the following error message (in orange)

```
Resetting the board
Failed to reset the board, upload failed
Resetting the board
Failed to reset the board, upload failed
Resetting the board
Failed to reset the board, upload failed

```

jandrassy, 2019-05-12:
tools.avrdude.upload.network_pattern is not set right. Do you try it with official AVR package? it will not work. Did you change the bootloader? 
https://github.com/jandrassy/ArduinoOTA#atmega-support
I recommend to use ["my_boards"](https://github.com/jandrassy/my_boards) custom hardware configuration to burn the special bootloader and to use the board with it.

peros550, 2019-05-12:
many thanks for your quick response. 
Is it mandatory to switch bootloader?

jandrassy, 2019-05-12:
> many thanks for your quick response.
> Is it mandatory to switch bootloader?

yes.

peros550, 2019-05-13:
I will try it after 2 weeks time and let you know. I will close the issue for now and if I need more help I will get back to you. 
Many thanks for your work,

closed 2019-05-13 by peros550

VitorCSoares, 2020-04-22:
Hi all, 

What was the solution for this? I think I'm having the exact same problem. I already burn a new bootloader.
Now I'm receiving the following log error message (after introducing the password)

**
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/arduinoOTA -address 192.168.1.177 -port 65280 -username arduino -password password -sketch C:\Users\Skine\AppData\Local\Temp\arduino_build_137800/OTA_Ethernet_v01_20200421.ino.bin -upload /sketch -b 
flag provided but not defined: -username
Usage of C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\arduinoOTA:
  -address string
    	The address of the board (default "localhost")
  -port string
    	The board needs to be listening on this port (default "80")
  -q	Quiet flag
  -reset string
    	Upload endpoint
  -sketch string
    	Sketch path
  -ssl string
    	SSL flag
  -sync string
    	Upload endpoint
  -sync_exp string
    	sync expected return code in format code:string
  -upload string
    	Upload endpoint
  -v	Verbose flag (default true)
  -version
    	Prints program version
**

jandrassy, 2020-04-22:
you have old version of ArduinoOTA. I see you have one copied in sketches folder's hardware subbofolder. delete it and/or install a new version of AVR boards package

VitorCSoares, 2020-04-22:
Hi,

  *   I removed the entire installation.
  *   I installed again everything (v1.8.112);
  *   I added ArduinoOTA from library management screen;

I already tried multiple approaches, so I would like to ask for your help in the next steps. I always have some difficulty with copy the platform.local.txt file step.

Thank you

jandrassy, 2020-04-22:
ok. not very efficient way, but it did install the latest upload tool because it is part of the AVR boards package installed with the IDE. now the AVR boards package is in `C:\Program Files (x86)\Arduino\hardware\arduino\avr` (but if you  update or downgrade the AVR boards package it will be installed in `%userprofile%\AppData\Local\Arduino15\packages\`)

VitorCSoares, 2020-04-22:
Yes, 
Now I have the installation of the tool again in the C:\Program Files (x86)\Arduino\hardware\arduino\avr.
I already have one Mega 2560 with  Optiboot bootloader.

I copied my_boards to the hardware folder and I can select the board ( Optiboot ) in the IDE. 

I can upload the sketch to my Mega 2560 board through USB but still the same problem when using OTA. 
_Resetting the board
Failed to reset the board, upload failed_

when introducing "password"

jandrassy, 2020-04-22:
the upload tool sometimes times-out even if the update is sucessful

VitorCSoares, 2020-04-22:
I got the solution. It was already presented by you multiple times. 
just copy the platform.local.txt to C:\Program Files (x86)\Arduino\hardware\arduino\avr. We will end with platform.txt and platform.local.txt in the same folder. 
Thank you :)

I did not understand your last comment about WiFiOTA.cpp

VitorCSoares, 2020-04-29:
Hi Jandrassy,

After multiple success tests now I'm having the following problem:

Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
Error flashing the sketch:
Error flashing the sketch:

Can you help me please?

jandrassy, 2020-04-29:
does it update the sketch? did you try to add the status print?

VitorCSoares, 2020-04-29:
No. The Sketch remains the same. 

Can you give a more detailed explanation of add the status Print? I did not get your point above in the previous message.

VitorCSoares, 2020-04-29:
I added the Status Code in the WiFiOTA.cpp.
I'm connecting the board via USB to check the Serial Port and I'm doing the update via IP address. 

I'm getting "Unauthorized" as Status. The prompt to insert the password is not appearing now. 

Thank you

jandrassy, 2020-04-29:
try to set the password in the command in platform.local.txt

VitorCSoares, 2020-04-29:
Hi, 

It's saying Unauthorized, but the problem seems to be related to the Ethernet Switch I'm using.
If I connect the board (via ethernet) to the same switch my computer is connected I can upload the code.
If I connect the board (via ethernet) to another switch that is connected to the switch my computer is connected I can't see the board Port nether reach it through Program upload.

Maybe one of the switches is blocking some specific port needed. 

Thank you

BartArIen, 2020-10-28:
HI, 

I have read and followed previous comments: 

1. added [my-boards](https://github.com/jandrassy/my_boards) to Arduino hardware
2. burned the Optiboot to the Arduino Mega 2560
3. copied platform.local.txt to  /Users/bart/Documents/Arduino/hardware/my_boards/avr so I have a platform.local.txt and a platform.txt file in the samen directory. 
4.  changed the password in platform.local.txt to the one in the sketchfile.

But I still get this error 
_Users/bart/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0/bin/arduinoOTA -address 192.168.0.150 -port {upload.network.port} -sketch /var/folders/xm/r4mp8nvn7xd7_mrcm7qfhq2r0000gn/T/arduino_build_187044/sketch_oct27a.ino.hex -upload {upload.network.endpoint_upload} -sync {upload.network.endpoint_sync} -reset {upload.network.endpoint_reset} -sync_exp {upload.network.sync_return} 
Resetting the board
Failed to reset the board, upload failed_

[image]

Is their a way I can check if the Optiboot loader is burned correctly? What other checks can I do to know what the problem can be?

jandrassy, 2020-10-28:
platform.local.txt goes to 'official' AVR boards package in IDE installation folder or in Arduino15 folder

BartArIen, 2020-10-28:
I have moved platform.local.txt to Arduino15, restarted the ide and arduino but still the same error. Any other suggestions i can try?

jandrassy, 2020-10-28:
> I have moved platform.local.txt to Arduino15, restarted the ide and arduino but still the same error. Any other suggestions i can try?

to Users/bart/Library/Arduino15/packages/arduino/hardware/avr?

BartArIen, 2020-10-29:
My mistake, in the correct folder as you specified it works, thnx!

VitorCSoares, 2020-12-18:
Hi @jandrassy. I realized when I do the upload of the code using the programmer I get a success message although the code remains the same. 
Did you faced this behaviour already?

Thank you

jandrassy, 2020-12-18:
https://github.com/jandrassy/ArduinoOTA#upload-returns-ok-but-the-sketch-is-not-replaced

VitorCSoares, 2020-12-18:
Does that mean I need to burn a new bootloader again? If I understood correctly this copy_flash_pages function is not present in the bootload I loaded in the past. 
Am I understanding this correctly? 

Sorry for the detailed question.

jandrassy, 2020-12-18:
sorry, the discussion in this issue is too long to read it again. how do you burn the bootloader?

VitorCSoares, 2020-12-18:
using your code extracted in May2020

jandrassy, 2020-12-18:
you meant my_boards. then the bootloader should be OK.
do you use watchdog in your sketch?

VitorCSoares, 2020-12-18:
no Watchdog. Can you point me how to add it?

jandrassy, 2020-12-19:
> no Watchdog. Can you point me how to add it?

I asked because it could reset the MCU before copy_flash_pages finished

try to burn the bootloader again. I have no better idea

matiasAS, 2021-06-15:
Hello @jandrassy

Sorry to ask again for the same. I have copied the platforms.local.txt file to Arduino / hardware / arduino / avr and got this result:
Connecting to board ... done
Uploading sketch ... done
Flashing sketch ... Error flashing the sketch
no protocol:

What I have not done because I understand well how to do it is burn the optiboot bootloader, could you help me please?

Extra information:
Operating system: Linux, Ubuntu 16
Arduino mega

Thank you
I am waiting for you to help me

jandrassy, 2021-06-16:
do you have the folder structure in sketchbook's hardware folder like this "hardware/my_boards/avr"?

matiasAS, 2021-06-16:
@jandrassy yes
I from Chile

structure folder:
[image]

list of boards
[image]

platform.local.txt into Arduino/hardware/arduino/avr:

[image]

result upload code:
[image]

regards

matiasAS, 2021-06-16:
@jandrassy no, sorry

my sketchbook is this:

[image]
regards

matiasAS, 2021-06-16:
after change sketchbook path as this:

[image]

i need copied and install libraries: arduinoota, arduinojson and pubsubclient (mqtt)

results:

[image]

@jandrassy It is the same original error, I think I am missing an intermediate step which is to burn the bootloader

can you help me with this please?

jandrassy, 2021-06-16:
your sketchbook folder is /home/Matias/Arduino and this should be in Preferences.
so the location of my_boards folder is right (/home/Matias/Arduino/hardware/my_boards)
do you have avr subfolder in my_boards?

matiasAS, 2021-06-16:
@jandrassy yes
I have cloned your repository https://github.com/jandrassy/my_boards.git and copied the my_boards folder inside the hardware folder

[image]

jandrassy, 2021-06-16:
what is the libraries folder?

matiasAS, 2021-06-16:
[image]

Arduino Mega 2560 (Optiboot) does not appear

jandrassy, 2021-06-17:
scroll the Tools Board menu down to the end.

matiasAS, 2021-06-17:
I have checked and nothing that says optiboot. It only appears if I replace the avr folder inside my_boards with the avr in hardware / arduino / avr, but if I do that I lose the programmers, none appear. Also it does not recognize the Arduino.h library

Do you speak Spanish?

matiasAS, 2021-06-17:
@jandrassy 
after update to 1.8.15:

[image]

matiasAS, 2021-06-17:
@jandrassy I have managed to get the optiboot boards to appear, but when loading the sketch I get the following result

[image]

matiasAS, 2021-06-17:
There is the possibility that we speak (writing because I do not speak English), by skype, whatsapp, email, more direct so as not to fill out this issue, I have more questions that I would like to ask you

jandrassy, 2021-06-17:
I don't speak Spanish
you have to burn the bootloader (Optiboot) to be able to upload with Optiboot

matiasAS, 2021-06-17:
@jandrassy  Can you please tell me how do I burn the bootloader? And another very important thing, can I go back to the original?

jandrassy, 2021-06-18:
https://www.arduino.cc/en/hacking/bootloader
https://www.arduino.cc/en/Tutorial/BuiltInExamples/ArduinoISP

matiasAS, 2021-06-18:
@jandrassy I see that the process is not so easy and since I have arduinos in production which is the ones I would like to load the bootloader to, I think it is not convenient for something to fail and run out of arduino. Thank you very much for your help, but I will keep it in mind.

matiasAS, 2021-06-18:
one more thing, I have an arduino mega that when loading the sketch appears timeout, that is a problem with the bootloader ?, and I have another that the usb port is not recognizable

matiasAS, 2021-06-18:
@jandrassy I took a risk with a mega arduino that I thought was dead and done, I see that it is not difficult. I give you a challenge: load the sketch binary from the browser in arduino mega like this for esp32: https://lastminuteengineers.com/esp32-ota-web-updater-arduino-ide/amp/

can it be done in e arduino mega?

jandrassy, 2021-06-18:
> @jandrassy I took a risk with a mega arduino that I thought was dead and done, I see that it is not difficult. I give you a challenge: load the sketch binary from the browser in arduino mega like [...]

see the Advanced examples. they show how to use InternalStorage to store and apply the update binary.

matiasAS, 2021-06-22:
@jandrassy When changing the bootloader, is compatibility with any library lost or is everything the same?

jandrassy, 2021-06-22:
> @jandrassy When changing the bootloader, is compatibility with any library lost or is everything the same?

no other library cares about bootloader. bootloader is only there to write the application to flash memory.

matiasAS, 2021-06-22:
> see the Advanced examples. they show how to use InternalStorage to store and apply the update binary.

from the browser with the arduino mega I mean this:

[image]

that's in an esp32, can it be done in arduino mega, upload the sketch from a url?

matiasAS, 2021-06-22:
> > @jandrassy When changing the bootloader, is compatibility with any library lost or is everything the same?
> 
> no other library cares about bootloader. bootloader is only there to write the application to flash memory.

ok, thanks

jandrassy, 2021-06-23:
> from the browser with the arduino mega I mean this:
>
> that's in an esp32, can it be done in arduino mega, upload the sketch from a url?

where do you see a problem?

matiasAS, 2021-06-23:
My question is because with at least 2 esp32 it has happened to me that when selecting the port (the ip of the esp32) to upload the sketch it does not appear even though it is connected. In order for it to appear again, I must connect it with the usb cable, upload the sketch and there it appears. So that there is that problem, it does not assure me that I can update the sketch if I require it remotely and that is why I think it is much better and safe to do it from a url like the image, because having a connection with the arduino / esp32, if there will be access to upload the sketch binary

do you understand me?

matiasAS, 2021-06-23:
I want to make a clarification, the esp32 that does not appear to me is connected to a wifi that is created in a raspberry as an access point in a subnet. The ip of the raspberry is 192.168.10.80 and the ip of the esp32 is 192.168.80.10, and that esp32 with the ip 192.168.80.10 does not appear in the list of ports

jandrassy, 2021-06-23:
https://github.com/jandrassy/ArduinoOTA#ota-upload-from-ide-without-network-port

matiasAS, 2021-06-25:
hello @jandrassy , I have burned the optiboot bootloader in the arduino mega, loaded my sketch, but the ip of the arduino does not appear in the ports to load the sketch remotely, how do I solve it?

[16 more replies]