JAndrassy/ArduinoOTA #7: Arduino MKR Zero with Ethernet Shield not visible

luke321, 2019-06-26:
Hello,

I am trying to get my MKR Zero OTA ready, the Ethernet Shield is working I can ping it.

I took the example code and changed it to a static IP.
I cannot see the MKR with a Bonjour Viewer und Windows 10, Bonjour installed with Itunes.
I also used a Network Monitor and could not see any Bonjour related trafic going out from the Arduino.

Do you have any suggestions?

jandrassy, 2019-06-26:
see here https://github.com/arduino/Arduino/issues/8408

luke321, 2019-06-26:
Thanks for the fast reply, but I already know that issue thread.
I cannot even see the Arduino in the Bonjour Browser app.

I also tried both Python and Arduino Versions suggested, but since the MKR is not visible in the Bonjour Browser I suspect the culprit elsewhere.

jandrassy, 2019-06-26:
do the examples of the Ethernet library work?
do you test ArduinoOTA with OTEthernet example?

luke321, 2019-06-26:
Yes i tested some examples they work, basic http server and also udp sender.
I am currently trying with the OTEthernet example.

jandrassy, 2019-06-26:
do the examples of the Ethernet library work modified to static IP?

luke321, 2019-06-26:
Yes they do, this makes it so puzzling for me. I will try another pc tomorrow but I only have Windows 10 available.

jandrassy, 2019-06-26:
it sometimes help to restart the IDE. 
on my Linux computer the mdns list shows the entry only after the IDE

luke321, 2019-07-16:
After some windows updates it works now, I actually have to start the IDE first and than start the Arduino while being connected via ethernet. It shows up immediately.

Thanks for the support and this great library!

closed 2019-07-16 by luke321

luke321, 2019-07-17:
Update:

If anyone encounters similar problems under Windows 10, make sure the network is not set to public, device discovery must be enabled.
And for the discovery to work reliably I had to disable all other network adapters and reboot.