JAndrassy/ArduinoOTA #8: MKR1000 OTA not showing?

TakeMeNL, 2019-07-17:
Hi there,
Saw the other topics, but i have a different scenario
So im using the 1.8.9 ide, latest WiFi101 (0.16.0) and latest firmware (19.6.1) and ArduinoOTA 1.0.1

Now i'm on windows, but i have multiple tools to check the mDNS. And it's just not showing up.
Even when i try the example OTA it's not showing. Also tried the WiFi101OTA lib not working either.

It did work couple off weeks ago, now it suddenly stopped working?
My default sketch is running: WiFi101, ArduinoOTA, PubSubClient (MQTT), WDTZero

Any more ideas how to trouble shoot this issue?
This is the only thing i see with packet sniffing
```
22:35:26.287088 f8:f0:05:ed:fb:5e > 01:00:5e:00:00:fb, ethertype IPv4 (0x0800), length 60: (tos 0xc0, ttl 1, id 0, offset 0, flags [none], proto IGMP (2), length 32, options (RA))
    10.10.1.90 > 224.0.0.251: igmp v2 report 224.0.0.251
22:35:26.584616 f8:f0:05:ed:fb:5e > 01:00:5e:00:00:fb, ethertype IPv4 (0x0800), length 60: (tos 0xc0, ttl 1, id 2, offset 0, flags [none], proto IGMP (2), length 32, options (RA))
    10.10.1.90 > 224.0.0.251: igmp v2 report 224.0.0.251
```

jandrassy, 2019-07-18:
you could try the original Arduino library WiFi101OTA

TakeMeNL, 2019-07-18:
`Also tried the WiFi101OTA lib not working either.`
Yes I have tried that. Not working

jandrassy, 2019-07-18:
you could try the beta version of IDE. the 1.9. it has new network port detection

TakeMeNL, 2019-07-18:
No no luck, because for some reason the arduino is not even trying to send the udp packets?
Idk, im just guessing and trying to figure out why it's not working.

jandrassy, 2019-07-18:
I didn't change the mdns part. I think it only listens for request to resolve the name

jandrassy, 2019-07-18:
did you read https://github.com/jandrassy/ArduinoOTA/issues/7 ?

TakeMeNL, 2019-07-18:
So apparently my windows machine had 2 static ips assigned to the interface. (Which i didn't know about...)
And it was broadcasting on the "wrong" network. i also reinstalled bonjour, but because that could be a issue to.
But it is working now, sorry for the inconvenience if any! Thank you any way :)

closed 2019-07-18 by TakeMeNL