JAndrassy/ArduinoOTA #10: Need help using ArduinoOTA with Arduino Mega

SandraWeddig, 2019-09-29:
Hi!

I have an Arduino Mega 2560 and would like to do OTA updates of my sketches.
As I understand, this is possible with you project.
Is there a step by step tutorial how to use it?

Thanks,
Sandra

jandrassy, 2019-09-29:
yes, it is possible
did you read this? https://github.com/jandrassy/ArduinoOTA#atmega-support

SandraWeddig, 2019-09-29:
Yes, I did read it. But I'm not sure, if I did understand and follow it all.

I did copy my_boards to /opt/arduino/hardware.

Then I did burn the bootloader Arduino Mega 2560 (Optiboot) to my board.
After this I did burn the network configuration (I use an ethernet shield) to the board and uploaded my sketch in which I did add "#include <ArduinoOTA.h>", "ArduinoOTA.begin(Ethernet.localIP(), "Arduino", "password", InternalStorage);" at the end of the setup function and "ArduinoOTA.poll();" in the loop().

There appears no network port in my Arduin IDE. I have disabled my client firewall and can ping the board.

jandrassy, 2019-09-29:
did you try the OTEthernet example?

SandraWeddig, 2019-09-29:
Yes, and it didn't worked till I changed to ArduinoIDE 1.8.3.

The bug https://github.com/arduino/Arduino/issues/8408 blocks network ports since version 1.8.5.

Thanks for your help. I will try the OTA update next (board is already under network ports) and give you feedback.

jandrassy, 2019-09-29:
but make sure you have AVR boards package at least version 1.6.21 installed for upload tool arduinoOTA version 1.2.

SandraWeddig, 2019-09-29:
I still fail at the password... with the default. :(

jandrassy, 2019-09-29:
copy here the upload command from the IDE console

SandraWeddig, 2019-09-29:
I only see "Resetting the board
Failed to reset the board, upload failed" in the command line, when I press the upload button from Arduin IDE. Now it works with version 1.8.9 also, but I doesn't differ when asking for the password.

jandrassy, 2019-09-29:
in Preferences check verbose mode for Upload

SandraWeddig, 2019-09-29:
The command is "/root/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0/bin/arduinoOTA -address 192.168.69.201 -port {upload.network.port} -sketch /tmp/arduino_build_393267/OTEthernet.ino.hex -upload {upload.network.endpoint_upload} -sync {upload.network.endpoint_sync} -reset {upload.network.endpoint_reset} -sync_exp {upload.network.sync_return} "

jandrassy, 2019-09-29:
upload with my_boards Mega

SandraWeddig, 2019-09-30:
I use the entry "Arduino Mega 2560 (Optiboot)".

jandrassy, 2019-09-30:
then try to copy platform.local.txt from extras/avr folder, next to platform.txt in the AVR boards package Packages are located in ~/.arduino15/packages/ on Linux and %userprofile%\AppData\Local\Arduino15\packages\ on Windows (AppData is a hidden folder).

SandraWeddig, 2019-09-30:
That did the trick :) 

Thanks, I can now upload via OTA.

closed 2019-09-30 by SandraWeddig

Kashyapkoshti, 2019-12-23:
> That did the trick :)
> 
> Thanks, I can now upload via OTA.

Can you please provide step by step procedure from beginning?