JAndrassy/ArduinoOTA #11: support for ESP32httpUpdate.h

mugginsjm, 2019-11-19:
Hi I have been using ESP32httpUpdate successfully over wifi.
Is there any way it can be made to work over ethernet?
thanks

jandrassy, 2019-11-19:
the best way is to make an Ethernet version of the ESP32 HTTPUpdate library. it uses the HTTPClient library, which uses the WiFiClient. replace WiFiClient with EthernetClient

closed 2019-11-19 by jandrassy

mugginsjm, 2019-11-19:
Thanks for that but II have tried every combination and nothing works.
I have a solid ethernet connection but on ota I get

wifi status is 255
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x400880c1 on core 1

Backtrace: 0x4008bbf0:0x3ffb1ba0 0x4008be21:0x3ffb1bc0 0x400880c1:0x3ffb1be0 0x40122ce2:0x3ffb1c20 0x40122ee2:0x3ffb1c40 0x40115344:0x3ffb1c60 0x401153ad:0x3ffb1c80 0x4011254a:0x3ffb1ca0 0x401113d4:0x3ffb1cc0 0x40114bc8:0x3ffb1d00 0x400d60a6:0x3ffb1d20 0x400d5b06:0x3ffb1d70 0x400d7335:0x3ffb1da0 0x400d73c5:0x3ffb1dd0 0x400d7493:0x3ffb1e10 0x400d5723:0x3ffb1e30 0x400d5942:0x3ffb1e90 0x400d1f1b:0x3ffb1f40 0x400d2052:0x3ffb1f70 0x400db145:0x3ffb1fb0 0x400883d5:0x3ffb1fd0

jandrassy, 2019-11-19:
it is really off topic here. this library is for upload from IDE and uses the Updater library of arduino-esp32 to apply the uploaded binary

mugginsjm, 2019-11-19:
OK
thanks anyway