JAndrassy/ArduinoOTA #13: Support for integration with WiFiEspAT.h

heinemannj, 2019-12-19:
I try to use the WIFI101_OTA example:

#include <SPI.h>
#include <WiFiEspAT.h>
#include <ArduinoOTA.h>

#include "arduino_secrets.h" 
.
.
.

In file included from C:\Users\A410882\AppData\Local\Temp\arduino_modified_sketch_74376\WiFi101_OTA.ino:23:0:

C:\Users\A410882\Documents\Arduino\libraries\ArduinoOTA\src/ArduinoOTA.h:109:2: error: #error "Network library not included or not supported"

 #error "Network library not included or not supported"

  ^~~~~

Multiple libraries were found for "SPI.h"
 Used: C:\Users\A410882\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.2\libraries\SPI
Multiple libraries were found for "WiFiEspAT.h"
 Used: C:\Users\A410882\Documents\Arduino\libraries\WiFiEspAT
Multiple libraries were found for "ArduinoOTA.h"
 Used: C:\Users\A410882\Documents\Arduino\libraries\ArduinoOTA
Using library SPI at version 1.0 in folder: C:\Users\A410882\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.2\libraries\SPI 
Using library WiFiEspAT at version 1.0.2 in folder: C:\Users\A410882\Documents\Arduino\libraries\WiFiEspAT 
Using library ArduinoOTA at version 1.0.1 in folder: C:\Users\A410882\Documents\Arduino\libraries\ArduinoOTA 
exit status 1
Error compiling for board ATmega2560.

jandrassy, 2019-12-19:
sorry I forgot to make a release. it is done now. you can download it from the Releases tab. in less then 24 hour it should be available in Library Manager

heinemannj, 2019-12-20:
Jepp - I confirm that the new version is available in Arduino IDE and compiling with WiFiEspAT lib is working as expected.

Many thanks.
But OTA is not working ... - I will open a new issue

closed 2019-12-20 by heinemannj