JAndrassy/ArduinoOTA #14: OTA with WiFiEspAT and optiboot8/optiboot_atmega2560.hex - No mdns record  and no OTA possible

heinemannj, 2019-12-20:
Device:				RobotDyn MEGA+WiFi R3 ATmega2560+ESP8266, flash 32MB, USB-TTL CH340G, Micro-USB

Original ESP8266 firmware:	ESP8266_NonOS_AT_V1.7.1_19_07_15

Arduino IDE:			1.8.10

#include <SPI.h>
#include <WiFiEspAT.h>
#include <ArduinoOTA.h>
#include <PubSubClient.h>

- Sketch and network conectivity is working as expected (e.g. ping, mqtt, ...)
- No arduino record available in mdns browser
- No network port in Arduino IDE
- No OTA

----------------------------------------------------------------------------
Installation of bootloader
----------------------------------------------------------------------------
Arduino\hardware\my_boards\avr/bootloaders/optiboot8/optiboot_atmega2560.hex
----------------------------------------------------------------------------

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega2560 -cstk500v1 -PCOM8 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xD6:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM8
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
[... 316 lines cut ...]
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "C:\Users\A410882\AppData\Local\Temp\arduino_build_510765/mqtt_esp8266_SO_Counter.ino.hex"
avrdude: writing flash (21464 bytes):

Writing | ################################################## | 100% 2.98s

avrdude: 21464 bytes of flash written
avrdude: verifying flash memory against C:\Users\A410882\AppData\Local\Temp\arduino_build_510765/mqtt_esp8266_SO_Counter.ino.hex:
avrdude: load data flash data from input file C:\Users\A410882\AppData\Local\Temp\arduino_build_510765/mqtt_esp8266_SO_Counter.ino.hex:
avrdude: input file C:\Users\A410882\AppData\Local\Temp\arduino_build_510765/mqtt_esp8266_SO_Counter.ino.hex contains 21464 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 2.19s

avrdude: verifying ...
avrdude: 21464 bytes of flash verified

avrdude done.  Thank you.

heinemannj, 2019-12-20:
[image]

heinemannj, 2019-12-20:
[image]

jandrassy, 2019-12-20:
I will add to library a SerialWiFiOTA example soon, but here it is
https://forum.arduino.cc/index.php?topic=653504.msg4412050#msg4412050

did you see this https://github.com/jandrassy/ArduinoOTA/issues/12 ?

heinemannj, 2019-12-20:
OK - using your example from
https://forum.arduino.cc/index.php?topic=653504.msg4412050#msg4412050
mdns issue by the usage of WiFiEspAT.h is resolved.

But next issue is follow up ...

heinemannj, 2019-12-20:
Sketch uses 16426 bytes (6%) of program storage space. Maximum is 261120 bytes.
Global variables use 1135 bytes (13%) of dynamic memory, leaving 7057 bytes for local variables. Maximum is 8192 bytes.
processing.app.debug.RunnerException: Unable to connect to 192.168.178.115
	at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:148)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2075)
	at java.lang.Thread.run(Thread.java:748)
Unable to connect to 192.168.178.115

jandrassy, 2019-12-20:
did you read the AVR part in README?
https://github.com/jandrassy/ArduinoOTA#atmega-support

heinemannj, 2019-12-20:
Yes

Arduino\hardware\my_boards\avr/bootloaders/optiboot8/optiboot_atmega2560.hex
is already installed.

Do I have to add additional cabeling between ESP and Arduino for required reset?

For the moment I only use the following board internal connects via DIP switches:

1-2-3-4 => ON
5-6-7-8 => OFF

RX03/TX03

jandrassy, 2019-12-20:
no. the Mega updates and resets itself. 

the command with ssh upload is not right. 
do you use my_boards or Mega Core?
if Mega Core, did you copy platform.local.txt from extras folder of library?

heinemannj, 2019-12-20:
[image]

jandrassy, 2019-12-20:
is it possible that you have old version of the arduinoOTA upload tool?

heinemannj, 2019-12-20:
Let me check if I can reproduce the isse on a second PC

For the moment I try to update from my WIN7 PC with a fresh installed Arduino IDE ...
But I can try from a WIN10 PC of my wife which have never heared about Arduino ...

jandrassy, 2019-12-21:
my_boards README 

> With Arduino IDE rename the platform.txt from my_boards to platform.local.txt and copy it next to platform.txt in the official board package.

heinemannj, 2019-12-21:
I can reproduce to issue on a WIN10 PC

AVR boards package at 1.8.2
arduinoOTA Upload tool at version 1.3.0

C:\Program Files (x86)\Arduino\hardware\arduino\avr\platform.local.txt
C:\Users\A410882\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\platform.local.txt

Content of both files:

## Create output (bin file)
recipe.objcopy.bin.pattern="{compiler.path}{compiler.elf2hex.cmd}" -O binary {compiler.elf2hex.extra_flags} "{build.path}/{build.project_name}.elf" "{build.path}/{build.project_name}.bin"

tools.avrdude.upload.network_pattern="{network_cmd}" -address {serial.port} -port 65280 -username arduino -password password -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b

tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.with_bootloader.hex:i"

heinemannj, 2019-12-21:
For the Password window if use: password
Is this the right one?

jandrassy, 2019-12-21:
here is -password password in tools.avrdude.upload.network_pattern. you can change it to "{network.password}" to use the password entered in IDE

heinemannj, 2019-12-21:
The error is remaining - even if I enter a wrong password:

processing.app.debug.RunnerException: Unable to connect to 192.168.178.115
at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:148)
at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
at processing.app.SketchController.upload(SketchController.java:732)
at processing.app.SketchController.exportApplet(SketchController.java:703)
at processing.app.Editor$UploadHandler.run(Editor.java:2075)
at java.lang.Thread.run(Thread.java:748)
Unable to connect to 192.168.178.115

heinemannj, 2019-12-21:
Can I test to upload manually via cmd command?
Any possibility for debugging?

heinemannj, 2019-12-22:
I try to upload manually via arduinoOTA, but also with no success:

C:\>
C:\>C:\Users\A410882\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0\bin\arduinoOTA -address 192.168.178.115 -port 65280 -username arduino -pass
word password -sketch C:\Users\A410882\Documents\Arduino\WiFiEspAT_OTA_1\WiFiEspAT_OTA_1.ino.arduino_mega.hex -upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch

C:\>

jandrassy, 2019-12-22:
I will try with IDE 1.8.10. I tested the new example two days ago, but only with Eclipse.

you can add a debug print in library to see what was the response (which the upload tool doesn't show). in WiFiOTA.cpp in function sendHttpResponse add `Serial.println(status);`

heinemannj, 2019-12-22:
08:50:34.443 -> Waiting for connection to WiFi
08:50:35.455 -> .
08:50:35.455 -> Connected to WiFi network.
08:50:35.455 -> SSID: area053
08:50:35.489 -> IP Address: 192.168.178.115
08:50:35.489 -> signal strength (RSSI):-59 dBm
08:50:35.489 -> 
08:50:35.489 -> Sketch uploaded via COM6
08:51:34.384 -> 
08:51:34.384 -> sendHttpResponse: OK

heinemannj, 2019-12-22:
After uploading via OTA Arduino mega onboard LED is blinking and Sketch is not loaded.
After uploading via serial COM6 sketch can be restored.

jandrassy, 2019-12-22:
> After uploading via OTA Arduino mega onboard LED is blinking and Sketch is not loaded.
> After uploading via serial COM6 sketch can be restored.

and what sketch do you upload?

jandrassy, 2019-12-22:
because bin file should be uploaded, not hex

heinemannj, 2019-12-22:
How to create the bin file? Only renaming was not successful

heinemannj, 2019-12-22:
OK - found the bin file in a temp-dir created by Sketch - Export compilied Binary:

C:\>C:\Users\A410882\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0\bin\arduinoOTA -address 192.168.178.115 -port 65280 -username arduino -pass
word password -sketch C:\Users\A410882\Documents\Arduino\mqtt_esp8266_SO_Counter\mqtt_esp8266_SO_Counter.ino.bin -upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch

12:33:31.271 -> Counter Pin 69: 0
12:38:06.730 -> 
12:38:06.730 -> WiFiOTAClass::pollServer: 
12:38:06.730 -> 1
12:38:06.730 -> 
12:38:24.506 -> 
12:38:24.506 -> WiFiOTAClass::sendHttpResponse: OK
12:38:25.925 -> 
12:38:26.195 -> Waiting for connection to WiFi
12:38:26.704 -> ......
12:38:29.203 -> Connected to WiFi network.
12:38:29.236 -> 
12:38:29.236 -> WiFiOTAClass::begin: 
12:38:29.236 -> 1941088448
12:38:29.236 -> Arduino
12:38:29.236 -> Basic YXJkdWlubzpwYXNzd29yZA==
12:38:29.236 -> 
12:38:29.236 -> SSID: area053
12:38:29.270 -> IP Address: 192.168.178.115
12:38:29.270 -> signal strength (RSSI):-60 dBm
12:38:29.270 -> 
12:38:29.270 -> Upload via OTA was successful
12:38:29.270 -> 
12:38:30.925 -> Attempting MQTT connection...connected
12:38:30.992 -> 
12:38:31.093 -> Counter Pin 47: 0

heinemannj, 2019-12-22:
Results:
a) Manually OTA upload was successful but with error code by arduinoOTA.exe
b) Upload via Arduino GUI was NOT successful

Using library ArduinoOTA at version 1.0.2 in folder: C:\Users\A410882\Documents\Arduino\libraries\ArduinoOTA 
Using library PubSubClient at version 2.7 in folder: C:\Users\A410882\Documents\Arduino\libraries\PubSubClient 
"C:\\Users\\A410882\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-size" -A "C:\\Users\\A410882\\AppData\\Local\\Temp\\arduino_build_413105/mqtt_esp8266_SO_Counter.ino.elf"
Sketch uses 22866 bytes (8%) of program storage space. Maximum is 261120 bytes.
Global variables use 2891 bytes (35%) of dynamic memory, leaving 5301 bytes for local variables. Maximum is 8192 bytes.
processing.app.debug.RunnerException: Unable to connect to 192.168.178.115
	at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:148)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2075)
	at java.lang.Thread.run(Thread.java:748)
Unable to connect to 192.168.178.115

Strange side effect - Only an .elf file was compiled and not the bin file:

"C:\\Users\\A410882\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-size" -A "C:\\Users\\A410882\\AppData\\Local\\Temp\\arduino_build_413105/mqtt_esp8266_SO_Counter.ino.elf"

jandrassy, 2019-12-22:
the settings in platform.local.txt should activate creation of the bin file
try to copy the Arduino/libraries/ArduinoOTA/extras/avr/platform.local.txt
next to avr package platform.txt

jandrassy, 2019-12-22:
solution is in my_boards new commit https://github.com/jandrassy/my_boards/commit/fc5eb4f3c46e60a0feca46f045e50a585de85e0c

you can remove the platform.local.txt from the official package. 
get platform.txt and programmers.txt from my my_boards GitHub repo and put it into your local copy of my_boards.
in my_boards in programmers.txt, configure the IP address and restart the IDE. In IDE select in Tools menu the "Arduino OTA (...)" programmer and use "Upload using programmer" from the Sketch menu in IDE.

heinemannj, 2019-12-23:
Juraj - I confirm that your patch under my_boards provides a suitable workarround!

Remaining strange side effects by using the GUI or CMD :

Global variables use 2879 bytes (35%) of dynamic memory, leaving 5313 bytes for local variables. Maximum is 8192 bytes.
C:\Users\A410882\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0\bin\arduinoOTA.exe -address 192.168.178.115 -port 65280 -username arduino -password password -sketch C:\Users\A410882\AppData\Local\Temp\arduino_build_352492\mqtt_esp8266_SO_Counter.ino.bin -upload /sketch -b 
Connecting to board ...  done
Uploading sketch ...  done
**Flashing sketch ... Error flashing the sketch
Error flashing the sketch**

But new sketches are definitely uploaded and active.

jandrassy, 2019-12-23:
set higher baud rate for communication with the AT firmware. 500000 baud works perfect. 250000 is a good value too (Mega doesn't like common baud rates lie 230400 or 460800)

and depending on your projects memory needs you can[ set a larger buffer](https://github.com/jandrassy/WiFiEspAT#buffering-and-buffers) for WiFiClient

heinemannj, 2019-12-24:
Changing to higher baud rate for communication with the AT firmware is not possible on my 	RobotDyn MEGA+WiFi R3 ATmega2560+ESP8266 ...
Larger buffer is not solving the issue.

I have reproduced the issue on an traditional Arduino Mega 2560 connected to an NodeMCU Lua Lolin V3 Module ESP8266 ESP-12E WIFI modul via serial1 and 500000 baud rate for communication.

closed 2020-01-03 by jandrassy