JAndrassy/ArduinoOTA #16: Hello,I have a problem while I try to write the program into flash

xieyi1393, 2020-01-12:
Log:
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/arduinoOTA -address 192.168.0.170 -port 65280 -username arduino -password password -sketch C:\Users\DELL\AppData\Local\Temp\arduino_build_272676/ArduinoOTA1.ino.bin -upload /sketch -b 
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
Error flashing the sketch

Why did it happend and how can I resolve it?I try to find Baidu and Google but there is no solution.

xieyi1393, 2020-01-12:
I'm using Arduino MEGA2560&Arduino Ethernet Shield.
The version of IDE:Arduino 1.8.10
The source code:
```
#include <EEPROM.h>

/*

 This example polls for sketch updates over Ethernet, sketches
 can be updated by selecting a network port from within
 the Arduino IDE: Tools -> Port -> Network Ports ...

 Circuit:
 * W5100, W5200 or W5500 Ethernet shield attached

 created 13 July 2010
 by dlf (Metodo2 srl)
 modified 31 May 2012
 by Tom Igoe
 modified 16 January 2017
 by Sandeep Mistry
 Ethernet version August 2018
 by Juraj Andrassy

 modified by Matt Elias for static IP, LED output & EEPROM reboot counter
 */
 
#include <SPI.h>
#include <Ethernet.h>
#include <ArduinoOTA.h>
//#define Serial SerialUSB
IPAddress ip(192, 168, 0, 170);
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, ip[2], ip[3] };
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  //while (!Serial);  // only needed for Leo?
  Serial.print("\n\nVer: 2\nReboots: ");
  int reboots;
  EEPROM.get(1, reboots);
  reboots++;
  EEPROM.put(1, reboots);
  Serial.println(reboots);

  // start the Ethernet connection:
  Serial.println("Initialize Ethernet");
  Ethernet.begin(mac, ip);
  Serial.println(Ethernet.localIP());
  // start the OTEthernet library with internal (flash) based storage
  ArduinoOTA.begin(Ethernet.localIP(), "Arduino","password", InternalStorage);
}

void loop() {
  // check for updates
  ArduinoOTA.poll();

  // add your normal loop code below ...
} 
```
Then,I change the configure from "arduino" to "Arduino":
```
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
Error flashing the sketch:
Error flashing the sketch:
```

jandrassy, 2020-01-12:
I have never seen "Error flashing the sketch:" with `:`. What does print it? With "Flashing sketch ...  done" arduinoOTA tool is done.

xieyi1393, 2020-01-21:
It print nothing.So I can't get any infomation from them.
I copyed all of the error log.

xieyi1393, 2020-01-21:
I used the Example OTA 'programmer' configuration in programmers.txt and copied The context of '/extras/avr/plantform.local.txt' to 'plantform.txt'

jandrassy, 2020-01-21:
and is the new sketch uploaded over OTA running?

xieyi1393, 2020-01-21:
I remember it doesn't work.

xieyi1393, 2020-01-21:
I will try again after I get home.It may take me about half an hour.

jandrassy, 2020-01-21:
after "Flashing sketch ...  done" the sketch is uploaded and applied and the arduinoOTA tool finished
The "Error flashing the sketch:" is a mystery. 

Add in WiFiOTA.cpp to `sendHttpResponse`
```
  Serial.print("OTA upload response: ");
  Serial.println(status);
```
to see if the upload was successful.

xieyi1393, 2020-01-21:
It doesn't work.I found there is nothing except the network Infomation(IP Address and so on)

jandrassy, 2020-01-21:
> It doesn't work.I found there is nothing except the network Infomation(IP Address and so on)

did you add the `Serial.println(status);` and it doesn't print anything?

xieyi1393, 2020-01-21:
> > It doesn't work.I found there is nothing except the network Infomation(IP Address and so on)
> 
> did you add the `Serial.println(status);` and it doesn't print anything?
Yes,I upload them by OTA but nothing displayed except network infomation.But it displayed if I upload by Serial Port(USB)

jandrassy, 2020-01-21:
did you add the print to WiFiOTA.cpp in the library in `sendHttpResponse` function?

xieyi1393, 2020-01-27:
Ok,I will add it and test the sketch later.

jandrassy, 2020-07-09:
so the "Error flashing the sketch:" is printed if IDE upload verbose mode is enabled. 
with ArduinoOTA 1.0.4 version the fail reason is printed after the semicolon

closed 2020-07-09 by jandrassy