JAndrassy/ArduinoOTA #18: Using 2MB of flash on MKR MEM shield

egjohnson17, 2020-03-28:
Hello,

Rather than use the example you give with the SD card to increase the program memory size for an application on the MKR1010, are you able to suggest how I might use just the 2MB of flash memory available on the MKR MEM Shield?
https://store.arduino.cc/usa/mkr-mem-shield

Thank you
Erik

jandrassy, 2020-03-29:
https://www.arduino.cc/en/Reference/WiFi101OTA
there is a SerialFlashStorage which this library inherited from WiFi101OTA library. It was implemented by Arduino. But for SD they created the SDU library to load the binary at boot. I don't know about a library to load the binary from SerialFlash at boot. It would be similar to SDU, but I don't have the hardware to make the library.

closed 2020-04-08 by jandrassy

egjohnson17, 2020-04-08:
Thank you for quick response, sorry I had been too busy to get back to this. I was able to get the OTA programming to work once I included an SD card into the MKR-MEM-Shield. I will see if I can find or write a library to load the binary from the SerialFlash at boot.

jandrassy, 2020-07-26:
arduino published SFU library