JAndrassy/ArduinoOTA #19: [SAMD21] undefined reference to `__text_start__'

xseignard, 2020-04-01:
Hello,
I'm trying to use your lib on a custom board, but for some reasons, it fails to compile due to the following error:

```
In function `OTAStorage::OTAStorage()':
OTAStorage.cpp:(.text._ZN10OTAStorageC2Ev+0x40): undefined reference to `__text_start__'
```

I tried various bootloader (microsoft uf2 and original arduino zero one).
Can the `variant.cpp` I use can cause this problem ? (I use this one https://github.com/sparkfun/Arduino_Boards/blob/master/sparkfun/samd/variants/SparkFun_SAMD_Mini/variant.cpp)

Any idea ?

Best regards

jandrassy, 2020-04-01:
__text_start__ is defined in the linker script. the linker script is in the variant folder. why do you use the Sparkfun SAMD package and to the Arduino one?

xseignard, 2020-04-01:
Here is my dev env:
- uf2 bootloader
- SparkFun_SAMD_Mini variant because it matches my pin and SERCOM usage
- platformio

For some reasons, `__text_start__` is missing in the linker script, see https://github.com/sparkfun/Arduino_Boards/blob/master/sparkfun/samd/variants/SparkFun_SAMD_Mini/linker_scripts/gcc/flash_with_bootloader.ld

Any idea how to overcome this ?

jandrassy, 2020-04-01:
`__text_start__` is the flash location of the running sketch so it is the location where the new binary will be copied after receiving.
add `__text_start__` to the ld script. see https://github.com/arduino/ArduinoCore-samd/blob/703b87e02d1ce61cd3dddcaa7c4f9ff50c537f80/variants/mkrzero/linker_scripts/gcc/flash_with_bootloader.ld#L68

and recommended reading https://arduino.stackexchange.com/questions/54484/adding-a-custom-board-to-the-arduino-ide/60660#60660

xseignard, 2020-04-02:
Hello Juraj,
Thanks for your help.
Since I don't use the arduino IDE, I created my own variant in platformio.
I added the `__text_start__` var in the linker script (see: https://github.com/xseignard/corpore-sano/blob/ota-ethernet/variants/Pioche/linker_scripts/gcc/flash_with_bootloader.ld#L68)

It compiles, but using `ArduinoOTA` makes my sketch not working anymore. It must be related to my limited understanding of your lib. I already use ethernet through UDP for other operations in my sketch (see: https://github.com/xseignard/corpore-sano/blob/ota-ethernet/src/main.ino#L69)

Is it possible that `ArduinoOTA` and my own code are conflicting ?

jandrassy, 2020-04-02:
I don't see any conflict. ArduinoOTA has a TCP server listening on port 65280 and UDP broadcast for MDNS

xseignard, 2020-04-02:
Thanks for your answer.
So what would be the problem then?
Without `ArduinoOTA` my code works as expected, but when I add `ArduinoOTA`([here](https://github.com/xseignard/corpore-sano/blob/ota-ethernet/src/main.ino#L7), [here](https://github.com/xseignard/corpore-sano/blob/ota-ethernet/src/main.ino#L71) and [here](https://github.com/xseignard/corpore-sano/blob/ota-ethernet/src/main.ino#L226))  it stops working, as in, no more UDP messages, and no debug on `SerialUSB`.
Any idea ?

jandrassy, 2020-04-03:
first, to debug add `while (!Serial);` after Serial.begin. then add more Serial.prints

closed 2020-04-08 by jandrassy