JAndrassy/ArduinoOTA #21: Upload with ArduinoOTA and OTEthernet to ESP32 pico Kit does not save in flash

gitariana, 2020-04-24:
Hy

Uploading with ArduinoOTA / OTEthernet does not save the transferred sketch in Flash.
I have no idea what the cause may be.

My steps:

1. ArduinoIDE 1.8.12 - Portable installed on D: \ ArduinoXXX \ arduino-1.8.12-2. Not started

2.ESP32 pico kit with SPI Wiznet W5500. Board package ESP32 1.0.4 is installed

3. My test program is running, with web server and MQTT

4. Delete the ArduinoOTA directory in
D: \ ArduinoXXX \ arduino-1.8.12-2 \ portable \ packages \ esp32 \ hardware \ esp32 \ 1.0.4 \ libraries \ ArduinoOTA

5. Text file platform.local.txt (RAW, 6 lines, 391 bytes content) copied next to platform.txt
D: \ ArduinoXXX \ arduino-1.8.12-2 \ portable \ packages \ esp32 \ hardware \ esp32 \ 1.0.4

6. New project with OTEthernet.ino from example created after
D: \ ArduinoXXX \ arduino-1.8.12-2 \ portable \ sketchbook \ ESP32 \ OTEthernet

7. MAC adjusted

8. Ethernet.init (33); inserted in setup () before Ethernet.begin (mac)

9. Start ArduinoIDE in D: \ ArduinoXXX \ arduino-1.8.12-2

10. Choose OTEthernet.ino, Board ESP32 pico Kit, Standard (not Minimal SPIFFS ..), COM-Port,
Compile, upload via COMx.

11. After uploading, the program runs and DHCP-IP is displayed as serial. So it works.

12. No Ethernet upload "Arduino-IP ..." is offered under Tools / Port

13. Under Sketch, select Export binding file

14. Binding file OTEthernet.ino.pico32.bin is stored in the project

------------------------------

? 1. Username Arduino must not be changed.
Is that intended?

? 2. Why can't upload be selected directly under Tools / Port: Arduino-IP ...?
What else am I missing?

? 3. When uploading via arduinoOTA.exe (see below) "Sketch uploaded successfully" is displayed.
Changes in the sketches do not arrive on the ESP32.
What can it be?

Thank you, Klaus

-----------------------------

Upload the bin file
D: \ ArduinoXXX \ arduino-1.8.12-2 \ portable \ sketchbook \ ESP32 \ OTEthernet> D: \ ArduinoXXX \ arduino-1.8.12-2 \ hardware \ tools \ avr \ bin \ arduinoOTA -address 192.168.2.54 -port 65280 -username arduino -password password -sketch OTEthernet.ino.pico32.bin -upload / data -b
Connecting to board ... done
Uploading sketch ... done
Flashing sketch ... done

Sketch uploaded successfully

jandrassy, 2020-04-24:
> ? 1. Username Arduino must not be changed.
> Is that intended?

the second parameter ('Arduino' in examples) is not the user name. it is the name for MDNS.
The fixed user name is how Arduino implemented it. And I accepted it as I enhanced the library  because it is not a multiuser system.

> ? 2. Why can't upload be selected directly under Tools / Port: Arduino-IP ...?
> What else am I missing?

Was the networl port shown with the esp32 ArduinoOTA library?

> ? 3. When uploading via arduinoOTA.exe (see below) "Sketch uploaded successfully" is displayed.
> Changes in the sketches do not arrive on the ESP32.
> What can it be?

Did it work with the esp32 ArduinoOTA library?

gitariana, 2020-04-24:
Thanks
ArduinoOTA from ESP32 , reboots continue

gitariana, 2020-04-24:
ArduinoOTA over WiFi runs perfect

jandrassy, 2020-04-24:
> 
> 
> ArduinoOTA over WiFi runs perfect

the network port too?

basic Ethernet library examples like WebServer work good?

gitariana, 2020-04-24:
Yes, work good with ethernet oder ethernet2 Libs

jandrassy, 2020-04-24:
`-upload /sketch`
not data. `data` is SPIFFS

gitariana, 2020-04-24:
Now it works with /sketch. Sorry, I have not seen /sketch.

? 2. What can be the problem here?

gitariana, 2020-04-24:
Thanks for your help.
? 2. What can be the problem here?
Is it necessary to start an mdns service?

jandrassy, 2020-04-25:
For esp32 ArduinoOTA library the IDE detects and shows the network port?

gitariana, 2020-04-25:
ArduinoOTA from ESP32 with ethernet, is not running, it reboots continue. So i don´t see the networkport. 
ArduinoOTA from ESP32 with Wifi work´s fine und the IDE show´s the networkport. Upload works fine

jandrassy, 2020-04-25:
could you try my ArduinoOTA library over esp32 WiFi, if it shows the network port?

gitariana, 2020-04-25:
yes, i test it, this morning

gitariana, 2020-04-25:
My Arduino IDEs now bring errors when translating the wifi projects. Nothing works anymore.
The 2nd IDE portable installation has probably changed the 1st IDE installation. Unfortunately, I can no longer quickly try using WiFi. Very bad for me and no idea what is lost.
I have to get it back. That's going to take time.
Is it possible to name your library as ArduinoOTA_2 or similar so that it can exist in parallel with the ArduinoOTA?
Thank you. I'll get back

jandrassy, 2020-04-25:
you can always reinstall the esp32 package in IDE Boards Manager.

if you want to rename the library, change the name of the folder, name of the ArduinoOTA.h and the entries in library.properties

gitariana, 2020-04-25:
I was able to solve my problem.
Wifi with ESP32 / ArduinoOTA goes and Networkport is displayed and the uploading goes directly from the IDE.
Wifi with Juraj / ArduinoOTA goes. Upload with Arduinoota.exe brings the following error message for Flashing. Networkport is not displayed.

D: \ ArduinoXXX \ arduino-1.8.12-2 \ portable \ sketchbook \ ESP32 \ SimpleWiFiServerOn> D: \ ArduinoXXX \ arduino-1.8.12-2 \ hardware \ tools \ avr \ bin \ arduinoOTA -address 192.168.2.26 -port 65280 -username arduino -password password -sketch SimpleWiFiServerOn.ino.pico32.bin -upload / sketch -b
Connecting to board ... done
Uploading sketch ... done
Flashing sketch ... Error flashing the sketch

gitariana, 2020-04-26:
By Wifi i mean webserver and MQTT

jandrassy, 2020-04-26:
and did you try my ArduinoOTA over WiFi?

You can add in WiFiOTA.cpp to `sendHttpResponse`
```
  Serial.print("OTA upload response: ");
  Serial.println(status);
```
to see if the upload result message

gitariana, 2020-04-26:
See her:
Wifi with ESP32 / ArduinoOTA goes and Networkport is displayed and the uploading goes directly from the IDE.
Wifi with Juraj / ArduinoOTA goes. Upload with Arduinoota.exe brings the following error message for Flashing. Networkport is not displayed.

D: \ ArduinoXXX \ arduino-1.8.12-2 \ portable \ sketchbook \ ESP32 \ SimpleWiFiServerOn> D: \ ArduinoXXX \ arduino-1.8.12-2 \ hardware \ tools \ avr \ bin \ arduinoOTA -address 192.168.2.26 -port 65280 -username arduino -password password -sketch SimpleWiFiServerOn.ino.pico32.bin -upload / sketch -b
Connecting to board ... done
Uploading sketch ... done
Flashing sketch ... Error flashing the sketch

jandrassy, 2020-04-26:
sorry I missed "WiFi" in "WiFi with..." :-)

try to print the result message in WiFi101OTA.cpp, so we see the error reason. because the upload tool doesn't print it.

gitariana, 2020-04-26:
WiFiOTA.cpp or WiFi101OTA.cpp?
I use ESP32/Wifi.h

jandrassy, 2020-04-26:
sorry, WiFiOTA.cpp in my ArduinoOTA library. WiFiOTA is the true main file of the library. (originally it was called WiFi101OTA and it was the main file of the Arduino WiFi101OTA library)

gitariana, 2020-04-26:
It is very interesting:
Serial-Out:
OTA upload response: OK
OTA upload response2: OK
then the ESP32 resets
---
the 2 modifys:
void WiFiOTAClass::sendHttpResponse(Client& client, int code, const char* status)
{
  Serial.print("OTA upload response: ");
  Serial.println(status);
   while (client.available()) {
    client.read();
  }
  client.print("HTTP/1.1 ");
  client.print(code);
  client.print(" ");
  client.println(status);
  client.println("Connection: close");
  client.println();
  delay(500);
  client.stop();
  Serial.print("OTA upload response2: ");
  Serial.println(status);
}
----------------
the DOS-Out:
D:\ArduinoXXX\arduino-1.8.12-2\portable\sketchbook\ESP32\SimpleWiFiServerOn>D:\ArduinoXXX\arduino-1.8.12-2\hardware\tools\avr\bin\arduinoOTA -address 192.168.2.26 -port 65280 -username arduino -password password -sketch SimpleWiFiServerOn.ino.pico32.bin -upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch

jandrassy, 2020-04-27:
`status` doesn't change in `sendHtppResponse`. it is send back to tool.

and is the new verison of the sketch applied?

gitariana, 2020-04-27:
Yes, Sorry. The sketch is in the Flash on ESP32 und runs.

gitariana, 2020-05-02:
Hello jandrassy,
my state:
Wifi: ESP32 / ArduinoOTA runs with network port,
Wifi: ESP32 / ArduinoOTA-Juraj runs with bin / exe, but error message,
Eth: ESP32 / ArduinoOTA is not running
Eth: ArduinoOTA-Juraj runs with bin / exe and OK message
Eth: ArduinoOTA-Juraj is not running with network port
It is so acceptable to me!
Do you have another idea?
Thank you
Klaus

hacki11, 2020-06-23:
Seems i have a similiar issue - my mega2560 sketch is uploaded successfully but the new code changes are not available on the device. shall i open a separate issue?

jandrassy, 2020-06-24:
> Seems i have a similiar issue - my mega2560 sketch is uploaded successfully but the new code changes are not available on the device. shall i open a separate issue?

esp32 way of applying the upload if very different then for Mega. so yes, but first check the closed issues.

closed 2020-12-09 by jandrassy

dyrvigk, 2021-01-25:
how to get it to work with ethernet2.h  dosent compile

jandrassy, 2021-01-26:
> how to get it to work with ethernet2.h dosent compile

why do you want to use Ethernet2?
you can modify ArduinoOTA.h to accept header guard define of Ethernet2

dyrvigk, 2021-01-26:
Because ethernet2 together with pubsubclient mqtt dosent make socket error
and ethernet.h does every  2 minute

i tried to change all the ethernet.h in arduinoOTA.h  to ethernet2.h but
still get a strange error

In file included from
C:\Users\dyrvi\AppData\Local\Temp\arduino_modified_sketch_941581\OTEthernet.ino:22:0:
C:\Users\dyrvi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA\src/ArduinoOTA.h:123:22:
error: 'WiFiServer' was not declared in this scope
 ArduinoOTAMdnsClass <WiFiServer, WiFiClient, WiFiUDP> ArduinoOTA;
                      ^
C:\Users\dyrvi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA\src/ArduinoOTA.h:123:34:
error: 'WiFiClient' was not declared in this scope
 ArduinoOTAMdnsClass <WiFiServer, WiFiClient, WiFiUDP> ArduinoOTA;
                                  ^
C:\Users\dyrvi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA\src/ArduinoOTA.h:123:53:
error: template argument 1 is invalid
 ArduinoOTAMdnsClass <WiFiServer, WiFiClient, WiFiUDP> ArduinoOTA;
                                                     ^
C:\Users\dyrvi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA\src/ArduinoOTA.h:123:53:
error: template argument 2 is invalid
C:\Users\dyrvi\AppData\Local\Temp\arduino_modified_sketch_941581\OTEthernet.ino:
In function 'void setup()':
OTEthernet:45:14: error: request for member 'begin' in 'ArduinoOTA', which
is of non-class type 'int'
   ArduinoOTA.begin(Ethernet.localIP(), "Arduino", "password",
InternalStorage);
              ^
C:\Users\dyrvi\AppData\Local\Temp\arduino_modified_sketch_941581\OTEthernet.ino:
In function 'void loop()':
OTEthernet:50:14: error: request for member 'poll' in 'ArduinoOTA', which
is of non-class type 'int'
   ArduinoOTA.poll();
              ^
exit status 1
request for member 'begin' in 'ArduinoOTA', which is of non-class type 'int'

kenn

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_-283181171020929079_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

jandrassy, 2021-01-26:
so Adafruit Ethernet2 has `ethernet_h` as include guard, which is same as Ethernet library so nothing needs to be changed in ArduinoOTA.h
but Ethernet2 doesn't have Udp.beginMulticast so you get an error. beginMulticast is used to propagete the network port, which can be turned off defining NO_OTA_PORT.
define NO_OTA_PORT before including ArduinoOTA.h:
```
#define NO_OTA_PORT
#include <ArduinoOTA.h>
```

dyrvigk, 2021-01-26:
i got it to work and i can upload sketch but flash fails

kenn

dyrvigk, 2021-01-26:
here is the test

C:\Users\dyrvi\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0\bin>arduinoOTA
-address 192.168.1.219 -port 65280 -username arduino -password password
-sketch w5500nilan.ino.nodemcu-32s.bin -upload /sketch
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch

C:\Users\dyrvi\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0\bin>arduinoOTA
-address 192.168.1.219 -port 65280 -username arduino -password password
-sketch w5500nilan.ino.nodemcu-32s.bin -upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch

jandrassy, 2021-01-26:
turn on verbose logging for upload in IDE preferences to see the response message

dyrvigk, 2021-01-26:
hi
i have gotten it to work i can upload my sketch after uploading serial your
OTEhernet  with only this addition to the code  Ethernet.init(5);
but i can only upload and flasch 1 time so there must be some
thing wrong with my code it uploads but fails the flash part i have put
verbose but where do i se the log

//#include <FS.h>          // this needs to be first, or it all crashes and
burns...
//#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson
#include "HardwareSerial.h"
//#include <SPIFFS.h>
//#include <WiFiClient.h>
#include <PubSubClientTools.h>
#include <PubSubClient.h>
#include <SPI.h>
#include <Ethernet.h>
//#include <ESPmDNS.h>
//#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <ModbusMaster.h>

//#include <WiFi.h>
//#include <ETH.h>
//#include <Ticker.h>
#define HOST "NilanGW-%s" // Change this to whatever you like.
#define MAXREGSIZE 28
#define SENDINTERVAL 5000 // normally set to 180000 milliseconds = 3
minutes. Define as you like
#define VENTSET 1003
#define RUNSET 1001
#define MODESET 1002
#define TEMPSET 1004
#define SELECTSET 500
#define USERFUNCSET 601
#define USERVENTSET 603
#define USERTIMESET 602
#define USERTEMPSET 604
#define COOLVENT 1101
#define COOLSET 1200
#define TEMPSET_T11 1700
#define TEMPSET_T12 1701

byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x9F
};
// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 1, 200);
IPAddress myDns(192, 168, 1, 1);

char chipid[12];
char mqtt_server[40] = "192.168.1.203";
char mqtt_port[6]  = "1883";
char mqtt_user[40] = "Kenn";
char mqtt_pass[40] = "Brazil4586";
EthernetServer server(80);
EthernetClient espClient;
PubSubClient mqttclient(espClient);
static long lastMsg = -SENDINTERVAL;
static int16_t rsbuffer[MAXREGSIZE];
ModbusMaster node;
char* usersetTopic1 = "ventilation/userset";
String req[4]; //operation, group, address, value
enum reqtypes
{
  reqtemp = 0,
  reqalarm,
  reqtime,
  reqcontrol,
[... 639 lines cut ...]
                text += (char)0x20; // replace degree sign with space
              } else {
                text += (char)(rsbuffer[i] >> 8);
              }
              mqname += (char *)name;
          }
          mqttclient.publish(mqname.c_str(), text.c_str());
        }
      }
      lastMsg = now;
    }
  }
  }

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_4477830655697899458_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

jandrassy, 2021-01-27:
with verbose output only one more line with one word is added to arduinoOTA tool output. it is the response code.it could be OK or Unauthorized or similar

dyrvigk, 2021-01-27:
OTEthernet sketch works fine here 2 times after each other
Sketch uses 303905 bytes (23%) of program storage space. Maximum is 1310720
bytes.
Global variables use 16484 bytes (5%) of dynamic memory, leaving 311196
bytes for local variables. Maximum is 327680 bytes.
C:\Users\dyrvi\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA
-address 192.168.1.224 -port 65280 -username arduino -password password
-sketch
C:\Users\dyrvi\AppData\Local\Temp\arduino_build_18336/OTEthernet.ino.bin
-upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
OK
Sketch uploaded successfully
Sketch uses 303905 bytes (23%) of program storage space. Maximum is 1310720
bytes.
Global variables use 16484 bytes (5%) of dynamic memory, leaving 311196
bytes for local variables. Maximum is 327680 bytes.
C:\Users\dyrvi\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA
-address 192.168.1.224 -port 65280 -username arduino -password password
-sketch
C:\Users\dyrvi\AppData\Local\Temp\arduino_build_18336/OTEthernet.ino.bin
-upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
OK
Sketch uploaded successfully

my sketch  works first time
Sketch uses 324549 bytes (24%) of program storage space. Maximum is 1310720
bytes.
Global variables use 19780 bytes (6%) of dynamic memory, leaving 307900
bytes for local variables. Maximum is 327680 bytes.
C:\Users\dyrvi\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA
-address 192.168.1.224 -port 65280 -username arduino -password password
-sketch
C:\Users\dyrvi\AppData\Local\Temp\arduino_build_527798/w5500nilan.ino.bin
-upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
OK
Sketch uploaded successfully

second time so there must be something wrong with my sketch
Sketch uses 324549 bytes (24%) of program storage space. Maximum is 1310720
bytes.
Global variables use 19780 bytes (6%) of dynamic memory, leaving 307900
bytes for local variables. Maximum is 327680 bytes.
C:\Users\dyrvi\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA
-address 192.168.1.224 -port 65280 -username arduino -password password
-sketch
C:\Users\dyrvi\AppData\Local\Temp\arduino_build_527798/w5500nilan.ino.bin
-upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch

now i cant even get the OTEthernet to flash again only via serial port then
it works again i don't understand what is the differens ok my code has
manny more things modbus mqtt hardware serial arduinojson and so
my code
https://pastebin.pl/view/e7d25179

kenn

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_8216925014352573596_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

jandrassy, 2021-01-27:
the successful receiving of your sketch first time is by the OTEthernet or by your sketch ?

dyrvigk, 2021-01-27:
by OTEthernet  only change to that code is the   Ethernet.init(5); under
setup

so the flashing of my code works with the OTEthernet  once  my code is on
the esp32 the upload works but not the flashing the code runs fine after
the first flash but it is not possible to flash again  via network

kenn

jandrassy, 2021-01-27:
try to add in setup() `ArduinoOTA.beforeApply([]() {yield();});`

dyrvigk, 2021-01-27:
same result

Sketch uses 324589 bytes (24%) of program storage space. Maximum is 1310720
bytes.
Global variables use 19780 bytes (6%) of dynamic memory, leaving 307900
bytes for local variables. Maximum is 327680 bytes.
C:\Users\dyrvi\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA
-address 192.168.1.224 -port 65280 -username arduino -password password
-sketch
C:\Users\dyrvi\AppData\Local\Temp\arduino_build_527798/w5500nilan.ino.bin
-upload /sketch -b
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

dyrvigk, 2021-01-29:
hi jandrassy 

i dident get futher i dont understand what the reason is for the flashing not working and i dont know how to debug it 

kenn

jandrassy, 2021-01-29:
I guess the upload tool times out. It was not tested with large binaries. I see people using curl for upload like here https://github.com/jandrassy/ArduinoOTA/issues/58

dyrvigk, 2021-01-29:
ok sounds funny because the OTEthernet scetch  uses 303905 bytes 23%
and my scetch uses   324589 bytes 24%  so not much differens and the
OTEthernet sketch can upload my sketch and flash once

kenn

jandrassy, 2021-01-29:
> ok sounds funny because the OTEthernet scetch uses 303905 bytes 23% and my scetch uses 324589 bytes 24% so not much differens and the OTEthernet sketch can upload my sketch and flash once 

I am out of ideas

dyrvigk, 2021-01-29:
do you know if there is any way off debugging maby via serial port

kenn

dyrvigk, 2021-01-29:
i have had the same code work with ota but with wifi standard arduinoota

kenn

dyrvigk, 2021-01-29:
but thanks anyway

i will se if i can get it to work

kenn

jandrassy, 2021-01-29:
add in WiFiOTA.cpp to `sendHttpResponse`
```
  Serial.print("OTA upload response: ");
  Serial.println(status);
```
to see the response message in Serial Monitor.