JAndrassy/ArduinoOTA #22: OTA using ethernet checksum failed

testter2, 2020-05-02:
i'm using atmega2560 connected to ethernet module to get binary file of esp32 then send the file through uart to esp32 to update its firmware.

I'm trying to get binary file data through ethernet module using HttpClient library but when the data stored in esp32 OTA partition, then i'm getting:
esp_image: Checksum failed . Calculated 0xad read 0x0

when i checked some of the received data from the binary file before being sent to esp32, i can see that some location have 0x0 instead of the right value it should be.

that's part of my code where i read the binary file and sent it to esp32.

```
EthernetClient transport;
HttpClient client(transport, SERVER, SERVER_PORT);
Serial.print("Check for update file ");
Serial.println(PATH);
client.get(PATH);
int statusCode = client.responseStatusCode();
Serial.print("Update status code: ");
Serial.println(statusCode);
if (statusCode != 200)
{
client.stop();
return;
}
long bin_length = client.contentLength();
if (bin_length == HttpClient::kNoContentLengthHeader) {
client.stop();
Serial.println("Server didn't provide Content-length header. Can't continue with update.");
return;
}
Serial.print("Server returned update file of size ");
Serial.print(bin_length);
Serial.println(" bytes");
ESPSerial.print(String(bin_length)); //send file size to ESP
delay(10000);
byte data_char;
int counter=0;
while (bin_length > 0)
{
if (!client.readBytes(&data_char, 1)) // reading a byte with timeout
break;
ESPSerial.write(data_char); //here send data to esp32
if(bin_length==0||bin_length==1||bin_length==2){Serial.print("last data=");Serial.println(data_char,HEX);}
counter++;
if(counter==1000)
{
Serial.println(bin_length);
counter=0;
delay(100);
}
bin_length--;
}
client.stop();
if (bin_length > 0)
{
Serial.print("Timeout downloading update file at ");
Serial.print(bin_length);
Serial.println(" bytes. Can't continue with update.");
return;
}
```

anyone could help me with this problem.

thanks in advance

jandrassy, 2020-05-03:
how did you get to this setup to update esp32 with Mega?

testter2, 2020-05-03:
> 
> 
> how did you get to this setup to update esp32 with Mega?

i'm really handling the communication between the esp32 and mega .so i can confirm that the data i received from ethernet on mega is sent to esp32.
i did the same for updating esp32 through serial Bluetooth and it worked perfectly.
here i used OTASketchDownload_SD.ino example and edited it so i send the data from i received from binary file to esp32 not storing them in internalStorage.

jandrassy, 2020-05-03:
but the esp32 can update itself and the OTASketchDownload(_SD) is too meant for Arduino to update itself.

testter2, 2020-05-03:
> 
> 
> but the esp32 can update itself and the OTASketchDownload(_SD) is too meant for Arduino to update itself.

i'm working on a project where ethernet module is connected to mega not to esp32 and i want to update esp32 firmware through ethernet.

jandrassy, 2020-05-03:
I would remove the delay()

closed 2020-05-03 by jandrassy

testter2, 2020-05-03:
> 
> 
> I would remove the delay()

if you mean delay(100).
this delay is coming after sending 1000 bytes so i could take time to store these 1000 bytes. in ESP32 OTA partition and not to miss any data coming from atmega.
why removing delay would would make a difference?
i removed that delay but still have the same problem.