JAndrassy/ArduinoOTA #25: not able to upload the first sketch atmega2560

golfredom, 2020-05-07:
trying to upload a example sketch on Arduino Mega 2650,  but i dont undertand the procedure. please help.
[image]

jandrassy, 2020-05-07:
select the network port and click the Upload button.
did you copy the platform.local.txt from extras folder next to platform.txt in boards packege folder?
did you burn the Optoboot bootoader to Mega?

golfredom, 2020-05-07:
Thanks for the reply, yes I did copy the file platform.local.txt

However I don’t know hoy to burn the optoboot boot loader to the mega. Can you explain please. 

Enviado desde mi dispositivo móvil!!!

jandrassy, 2020-05-07:
https://www.arduino.cc/en/Tutorial/ArduinoISP

closed 2020-05-13 by jandrassy

peros550, 2020-05-23:
@jandrassy I need to ask a question. 

I'm trying to change the bootloader to an Arduino Mega 2560. I have prepared an Arduino UNO as an ISP programmer, I have made the correct connections between the two boards (Pogrammer: UNO, Target: Mega2560) 

I am ready to burn the new bootloader as per the instructions [here](https://www.arduino.cc/en/Tutorial/ArduinoISP) and [here ](https://arduino.stackexchange.com/questions/473/how-do-i-burn-the-bootloader)

I have seen in your instructions that you mention: 

"You can download my boards definitions and use it to burn the modified Optiboot and to upload sketches to your Mega over USB and over network."

But I really don't know how to use your "my boards". This part is not clear to me. I think I need to place those files somewhere (probably under C:\Program Files (x86)\Arduino\hardware) , but don't know If I need to replace existing files in Arduino subfolder or put them on a separate folder. 

Could you please clarify?

jandrassy, 2020-05-23:
look at the picture
https://github.com/jandrassy/my_boards/blob/master/README.MD