JAndrassy/ArduinoOTA #26: Support for Arduino DUE (Atmel SAM3X8E ARM Cortex-M3)

anklimov, 2020-05-23:
Hi!
Is it possible to extend OTA support for Atmel SAM3X8E ARM Cortex-M3 ?
The architecture is quite similar with already supported platforms and from the first look, it should be possible to do, but lays behind my skills
I will appreciated for any kind help or advises

jandrassy, 2020-05-23:
best example of flash operations is for any MCU architecture is the eeprom emulation library.
https://github.com/PhiJu5/DueFlashStorage

details are in the datasheet of the MCU.
https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-11057-32-bit-Cortex-M3-Microcontroller-SAM3X-SAM3A_Datasheet.pdf

InternalStorage.cpp should be enhanced for ARM MCUs

anklimov, 2020-05-24:
Thank you for direction for investigations
I've just added SAM3X8E to Internal storage
Currently it was compiled successfully.
But I didn't realize how to test it)
It looks like every OTA flashing tools are "ESP oriented" and the usual DUE flasher "bossac" didn't understand OTA. 
espota.py as well is not working
May you point me to appropriate tool? 
Isn't yet integrated natively to Platformio framework?

jandrassy, 2020-05-24:
for Arduino IDE setup arduinoOTA in platform.local.txt similar to this
https://github.com/jandrassy/ArduinoOTA/blob/master/extras/avr/platform.local.txt

gcharles81, 2020-08-23:
@anklimov 

Hello I am testing your fork and managed to get it working with Arduino IDE & Command line 
I encountered an issue only with .bin file that has a size of 165kb while it is working fine with a .bin file of 30kb

 below is serial output for one successful and one fail upload

**_Successful_**
Erase flash
======================================================================================================================
Received bytes:30000
Flashing bytes:30000
From:C0000
To:80000
PageSize:256

**_Failed_**

Erase flash
================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================
Received bytes:130855

anklimov, 2020-08-23:
@gcharles81 - thanks for testing, but result is little bit unexpected. In my case, lib working against 139680 bytes binary
The maximum firmware size is 50% of flash. Did you use SAM3X8E with 512K flash?

gcharles81, 2020-08-24:
@anklimov  Thanks its also strange to me Yes I use SAM3X8E 

Today I try ti did a bit deeper and modified in WiFiOTA.cpp the sendHttpResponse function and I add there a   Serial.println(status) ; So I get a better understanding what is failing 

The message I get on the Serial monitor after the fail is Payload size wrong , I have to google a bit maybe I find something 

Open flash for write

Erase flash
================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================
Received bytes:126759
Payload size wrong

gcharles81, 2020-08-24:
Solved Issue by using Baud rate of 250000 instead of 115200 , now I can upload sucsessfully

anklimov, 2020-08-30:
@gcharles81 - as far as firmware upload going through LAN connection, not thought serial, that is not looks like root cause of failures. May you share more details?

gcharles81, 2020-08-30:
@anklimov  - In my case I am using ESP-01 connected via serial to my Arduino DUE and using WiFiEspAT library that is why was failing with 115200 Baud.

Why it was failing I do not know maybe some sort of  timeout. Thanks Charles

StybiPL, 2021-02-14:
@anklimov Your fork works perfectly on Ethernet Shield V1

If you want to upload sketches directly from Arduino IDE place my _platform.local.txt_ to PREFERENCES_FOLDER\packages\arduino\hardware\sam\1.6.12

`# This configuration file supports the general ArduinoOTA library https://github.com/jandrassy/ArduinoOTA

## arduinoOTA as programmer. add entries with {ip} into programmers.txt
tools.bossac.network_cmd={runtime.tools.arduinoOTA.path}/bin/arduinoOTA
tools.bossac.upload.network_pattern="{network_cmd}" -address {serial.port} -port 65280 -username arduino -password password -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b`

nanitook, 2022-05-08:
Any updates on this? Is @anklimov 's fork working for OTA updates from http servers?

anklimov, 2022-05-08:
@nanitook - my fork have basic logic, derived from  @JAndrassy ' s project - sketch act as http server, not client (you need POST firmware to controller using external routine)
Only differences of my fork:
- Working with DUE
- Possible to re-use http server for extend OTA API
- Some abstractions added to GET/POST files/another persistent data sharing same OTA API

nanitook, 2022-05-08:
> sketch act as http server, not client (you need POST firmware to controller using external routine)

I'm talking about something like [this](https://github.com/JAndrassy/ArduinoOTA/blob/596027bb3e4b81d4de35200ec1489ef2f207d471/examples/Advanced/OTASketchDownloadWifi/OTASketchDownloadWifi.ino). Downloading the new sketch from a remote http server and make the upgrade. If that's not possible, the only way to upgrade using OTA is using the arduino IDE?

anklimov, 2022-05-11:
Hello 
The provided example is quite relevant. 
It should be possible to use InternalStorage to upgrade FW, retrieved from any source. 
NB - the example using ESP http(s) libs. If you will  migrate this on another platform - some effort required here and not every platform is supported https. 
---
Andrey Klimov