JAndrassy/ArduinoOTA #28: help for macOS

peros550, 2020-06-06:
I worked on a windows machine and I was able to change bootloader on an Arduino mega. 
I have a Mac computer which I mainly use and was able to OTA send firmware to the Mega board without any modifications on the IDE. 

However, after changing bootloader, I cannot flash by serial port. I know I need to use the "my_board" definitions, but any idea where should I place it on a macOS computer?

many thanks

jandrassy, 2020-06-06:
Hi. you place it the Sketchbook folder's `hardware` subfolder. The Sketchbbok folder is configured in IDE Preferences.

closed 2020-06-06 by jandrassy

peros550, 2020-06-06:
if there is no hardware subfolder there I should create one right?

peros550, 2020-06-06:
thank you! I created the folder hardware because it wasn't there and it worked perfectly fine.