JAndrassy/ArduinoOTA #29: Info: PlatformIO,  Arduino Mega,  build & upload OTA

pantata, 2020-06-08:
Tested for macOS.
1. add to platformio.ini
```
 extra_scripts = post:extra_scripts.py
 upload_protocol = custom
 upload_flags =
   http://arduino:password@ip_address:65280/sketch
```
2. write extra_scripts.py next to platform.ini
```
Import("env", "projenv")

env.Replace(    
    UPLOADCMD="curl --request POST --data-binary @\"$BUILD_DIR\"${PROGNAME}.bin $UPLOADERFLAGS"
)

env.AddPostAction(
    "$BUILD_DIR/${PROGNAME}.elf",
    env.VerboseAction(" ".join([
        "$OBJCOPY", "-O", "binary",
        "\"$BUILD_DIR\"/${PROGNAME}.elf", "\"$BUILD_DIR\"${PROGNAME}.bin"
    ]), "Building $BUILD_DIR/${PROGNAME}.bin")
)
```

LS.

jandrassy, 2020-06-08:
platform.io doesn't process [platform.local.txt](https://github.com/jandrassy/ArduinoOTA/blob/master/extras/avr/platform.local.txt) copied from extras folder next to platform.txt?

pantata, 2020-06-08:
Zdravím,

Díky za tip, zkusím to upravit v tom souboru, jestli to zafunguje taky. 
Stejně není k dispozici arduinoOTA tool (nebo jsem ho nikde v instalaci frameworks Arduino nenašel), takže je potřeba uploadovat jinak a curl je v systému.

LS.

jandrassy, 2020-06-09:
nieco ako
/packages/arduino/tools/arduinoOTA/1.3.0/bin

pantata, 2020-06-09:
Neni tam. A jak se divam do kodu build scriptu main.py a arduino.py, tak mi pripada, ze to konfigurace desek a programatoru bere z vlastnich json souboru a ne z boards.txt.

V kazdem pripade, to co jsem postnul nahore funguje, jeste to asi pujde zjednodusit pomoci UPLOAD_COMMAND primo v platformio.ini projektu (vyzkousim vecer).

jandrassy, 2020-06-26:
to summarize in english, platformio doesn't use the platform.txt file of Arduino boards packages

maxgerhardt, 2020-06-29:
For reference: I did build a reference PlatformIO + ESP32 + Ethernet (W5500) + OTA demo here: https://github.com/maxgerhardt/pio-esp32-ethernet-ota 

For ESP32 it also seems that the Ethernet library needs some extra fixes ([ref](https://github.com/arduino-libraries/Ethernet/pull/107)) in the latest version, which I have included my version. 

Also @jandrassy, where's the magical `arduinoOTA` binary? 

https://github.com/jandrassy/ArduinoOTA/blob/ffb0d462d8e961ed52920fa8775cda3e0d3078c7/extras/esp32/platform.local.txt#L4

I coudln't find it in https://github.com/espressif/arduino-esp32/. 

Calling into `curl` is not the best way yet to add PlatformIO integration, since we cannot assume that to be installed (especially not on Windows). A general Python script would be better. I guess a script using python-requests could be built that simply executes the `POST` request like CURL does?

jandrassy, 2020-06-29:
> Also @jandrassy, where's the magical `arduinoOTA` binary?
> 
> https://github.com/jandrassy/ArduinoOTA/blob/ffb0d462d8e961ed52920fa8775cda3e0d3078c7/extras/esp32/platform.local.txt#L4
> 
> I coudln't find it in https://github.com/espressif/arduino-esp32/.

in ArduinoIDE and in Eclipse the upload tool is installed with Arduino AVR and SAMD boards packages. the AVR boards package is bundled with IDE so the tool is always there.

jandrassy, 2020-08-07:
this library was on my request removed from PlatformIO library register

closed 2020-08-07 by jandrassy

pscott-ams, 2022-03-18:
> For reference: I did build a reference PlatformIO + ESP32 + Ethernet (W5500) + OTA demo here: https://github.com/maxgerhardt/pio-esp32-ethernet-ota

@maxgerhardt I've posted an issue on your github page about not being able to connect to the IP/port when trying to upload with OTA. Could you take a look at it please?