JAndrassy/ArduinoOTA #31: arduinoOTA.exe seems to upload fine but esp returns Checksum failed

bnjroos, 2020-06-14:
Hi,

When I update my firmware using arduinoOTA, the update seems to work fine:
```
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done

Sketch uploaded successfully
```
however the firmware is not updated an the esp returns the following error:
`E (23612) esp_image: Checksum failed. Calculated 0x66 read 0x0`

Any idea what could be the problem ? I build the firmware.bin using platform.io

TIA

jandrassy, 2020-06-14:
Hi. espota over WiFi with esp8266 ArduinoOTA library works?

bnjroos, 2020-06-14:
Sorry I have not presiced my setup, I use esp32 with Ethernet library and
wiz5500.

jandrassy, 2020-06-14:
https://github.com/espressif/arduino-esp32/issues/3956

paveljenicek, 2020-08-08:
Hello Juraj,

same problem here - getting:

esp_image: Checksum failed. Calculated 0x88 read 0x0 

OTA over WiFi with ArduinoOTA library works fine.

Pavel

paveljenicek, 2020-08-08:
My fault, I forgot to add -b at the end of the flash command (Windows):

arduinoOTA.exe -address 192.168.0.79 -port 65280 -username arduino -password password -sketch mysketch.esp32.bin -upload /sketch -b

Now it works fine.

pj