JAndrassy/ArduinoOTA #32: Error flashing the sketch at Arduino Mega 2560

SandraWeddig, 2020-06-21:
Hi!

I get following message when I try to update my Arduino Mega 2560 via ArduinoIDE (using programmer with IP). The firewall is off.

`/opt/arduino-1.8.13/hardware/tools/avr/bin/arduinoOTA -address 192.168.2.101 -port 65280 -username arduino -password password -sketch /tmp/arduino_build_28229/Sketch_Lichtschalter_Arduino_Uno.ino.bin -upload /sketch -b 
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
Error flashing the sketch`

What can I do?

peros550, 2020-06-21:
I’m having the same issue. Most of the times it only takes a reboot at the board and the problem goes away. Then I can upload ok, but next I face again the same issue and need another reboot

jandrassy, 2020-06-21:
Connect the board to computer and add in WiFiOTA.cpp to `sendHttpResponse`
```
  Serial.print("OTA upload response: ");
  Serial.println(status);
```
to see if the response message.

hacki11, 2020-06-21:
I did a tcpdump and got Unauthorized.

`sudo tcpdump -A -s 0 'tcp port 65280 and (((ip[2:2] - ((ip[0]&0xf)<<2)) - ((tcp[12]&0xf0)>>2)) != 0)'`

hacki11, 2020-06-21:
I thought, the first string is the username but it is not!
username will always be "arduino" and password is the second string.
`ArduinoOTA.begin(Ethernet.localIP(), "not-a-username", "password", InternalStorage);`

jandrassy, 2020-06-21:
the second parameter of begin() is the name to show in Ports menu

SandraWeddig, 2020-06-24:
I tried the Ethernet example and it works. So there is something bad in my code.

Thanks for your help!

closed 2020-06-24 by SandraWeddig