JAndrassy/ArduinoOTA #34: Uploading large program (~70kb) fails using httpclient

Park0, 2020-06-26:
I have a mega2560 and a w5500. Those are connected to MQTT. So i can trigger a update using a MQTT command. I used the code from the advanced example using httpclient. 
It works fine if i flash a small program. But if i try to program i made ~71kb it fails to boot afterwards. I can upload the bin file using avrdude and it works/boots just fine.

jandrassy, 2020-06-26:
Hi. did you try the basic upload example OTEthernet?

Park0, 2020-06-26:
Ill try them both and report back thx

Park0, 2020-06-26:
I can upload it using the example. Ill have to try rewriting the code exclude the httpclient. It will take more time to try.

Park0, 2020-06-26:
I think i found the issue. I copy/pasted the example too much :D, having a new instance of EthernetClient for the download messed everything up i guess. 
Thanks for your time making this nice library

closed 2020-06-26 by Park0

Park0, 2020-06-28:
@jandrassy 
I have to correct myself. The main cause was the length in the httpclient. I fixed it here: https://github.com/arduino-libraries/ArduinoHttpClient/pull/90
I made a example for me to test it. I also added as a PR https://github.com/jandrassy/ArduinoOTA/pull/36 so i can be of use to someone else. It's a bit quick & dirty bit it will show how it works.

jandrassy, 2020-06-28:
yes, sorry, now I remembered about the problem with int for content-length