JAndrassy/ArduinoOTA #35: ESP32: OTA with IP address (no hostname) is failing

Miq1, 2020-06-26:
I got a setup here where the ESP32 is connected by Ethernet and is put behind a firewall, that does port forwarding of port 3232 to the ESP. So I cannot use the host name as put into the ArduinoOTA.begin() call, but only the firewall's IP address. When trying to do an OTA update, the invitation packets seem to be sent, but never answered:
```
Uploading .pio\build\az-delivery-devkit-v4\firmware.bin
15:17:59 [DEBUG]: Options: {'timeout': 10, 'esp_ip': '192.168.178.77', 'host_port': 23137, 'image': '.pio\\build\\az-delivery-devkit-v4\\firmware.bin', 'host_ip': '0.0.0.0', 'auth': 'xxxxx, 'esp_port': 3232, 'spiffs': False, 'debug': True, 'progress': True}
15:17:59 [INFO]: Starting on 0.0.0.0:23137
15:17:59 [INFO]: Upload size: 402944
Sending invitation to 192.168.178.77 ..........
15:19:39 [ERROR]: No response from the ESP
*** [upload] Error 1
```
Is it possible that he ESP is discarding the packets due to the missing host name?

Is there a way to get around that?

My development IDE is VSCode/Platformio, btw.

jandrassy, 2020-06-26:
use the public IP of the 'firewall', not the local IP behind it

Miq1, 2020-06-26:
> 
> 
> use the public IP of the 'firewall', not the local IP behind it

That is in fact what I am doing. The IP of the ESP is 192.168.199.40, as compared to that of the firewall/gateway that you find in the log I enclosed. The router acting as firewall is spanning the 192.168.199.0/24 subnet.

jandrassy, 2020-06-26:
doesn't this platformio upload do a espota.py way, not the arduinoOTA tool way upload

Miq1, 2020-06-26:
If I only knew the difference... I installed your library from within PlatformIO, where it is offered as part of the Arduino environment (as compared to the ESP-IDF environment), so I was assuming it would be integrated. There is an esptool.py being used obviously, but that is about all I know about it.

jandrassy, 2020-06-26:
https://github.com/jandrassy/ArduinoOTA#esp8266-and-esp32-support

Miq1, 2020-06-26:
I have read that, thanks, but I think this is referring to using the Arduino IDE? I do not use that, but PlatformIO only with the Arduino _environment_. All the paths and commands mentioned in your linked section are belonging to the Arduino IDE, right?

Miq1, 2020-06-26:
Well, your second hint to the ``#29`` ``extra_scripts.py`` method seems to have other problems with my PlatformIO environment:
```
Building C:\Users\Micha\Documents\PlatformIO\Projects\RS485-Bridge\.pio\build\az-delivery-devkit-v4/firmware.bin
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument operation: invalid choice: 'binary' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'version')
*** [.pio\build\az-delivery-devkit-v4\firmware.elf] Error 2
```

Rats. :(

jandrassy, 2020-06-26:
platformio is not Arduino. they change things and lose compatibility. I don't use platformio and I don't plan to support it

closed 2020-06-26 by jandrassy