JAndrassy/ArduinoOTA #40: U_SPIFFS was not declared in this scope

markg85, 2020-08-01:
Hi,

I'm using an ESP-12e (8266) and am getting this error:

```
/home/mark/.platformio/lib/ArduinoOTA/src/InternalStorageESP.cpp: In member function 'virtual int InternalStorageESPClass::open(int, uint8_t)':
/home/mark/.platformio/lib/ArduinoOTA/src/InternalStorageESP.cpp:35:56: error: 'U_SPIFFS' was not declared in this scope
   return Update.begin(length, command == 0 ? U_FLASH : U_SPIFFS);
                                                        ^
/home/mark/.platformio/lib/ArduinoOTA/src/InternalStorageESP.cpp:36:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
*** [.pio/build/esp12e/lib391/ArduinoOTA/InternalStorageESP.cpp.o] Error 1
```

And with some googling i found this pull request:
https://github.com/awtrix/AWTRIX2.0-Controller/pull/15

Which solves that very issue in some other library (by replacing U_SPIFFS with U_FS).

If i do that in ArduinoOTA/src/InternalStorageESP.cpp the it does seems to compile just fine for me.

I'm not yet at the point where i can really test if this works though.. I'll update this post if i have confirmed a working flash with that flag.

Cheers,
Mark

jandrassy, 2020-08-01:
https://github.com/jandrassy/ArduinoOTA/issues/23

markg85, 2020-08-01:
Lol, not cool...
I've had these kind of issues more in the arduino world. Where i found a bug in a library, made up a possible workaround which is then already in the library, just not released yet. Quite annoying...

I installed that library today and with the latest version!

Could you consider a release?

jandrassy, 2020-08-01:
sorry, the fix is lost somehow

closed 2020-08-01 by jandrassy

markg85, 2020-08-01:
> sorry, the fix is lost somehow
> edit: I forgot about the PR #24 and deleted it accidentally with a git force push of my local repo
> I made a new commit with the fix

Please try to see this as a lesson.
Never! NEVER! use force push! If you need to do that you already have a problem in your git repository.

But i'm glad you added it again :)

jandrassy, 2020-08-02:
I used a force push because I didn't know why it didn't wan't to commit. I forgot about the PR because in my repos I am usually alone :-)

markg85, 2020-08-02:
> I used a force push because I didn't know why it didn't wan't to commit. I forgot about the PR because in my repos I am usually alone :-)

Ha :) yeah, i understand.
Just keep force push to star wars games from now on ;)