JAndrassy/ArduinoOTA #41: How to download a firmware update without losing WIFI connection?

markg85, 2020-08-01:
Hi,

I'm trying to use this advanced example: https://github.com/jandrassy/ArduinoOTA/tree/master/examples/Advanced/OTASketchDownload

I, at first, thought it was working. And - stupidly - updated all my custom lights (about 5 of them) to use this. Then to figure out that the download isn't finishing. It's not because the Wifi antenna apparently gets blocked or something like that? It works if my firmware is small enough, say around 250 KiB. But it's ~300 KiB.

The code in that example is blocking. It downloads the file in one while loop.
Any help or idea on how to make this async would be awesome!

Cheers,
Mark

jandrassy, 2020-08-02:
what board?

markg85, 2020-08-02:
It's the ESP-12e board from espressif. So not an arduino board with all kinds of extras soldered onto it.

I "assumed" ethernet meant i needed an RJ45 shield for wired ethernet? And as i already had a "WiFiClient" instance i used that instead. It seemed logical to me, but it's all based on assumptions. Did i make the wrong assumptions there?

I was using the ArduinoHttpClient. Or rather, some lights in my house still are but can't be updated now because of it :(

Having said all of this, i have discovered "[ESPhttpUpdate](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h)" which seems far easier to use and just does exactly what i need it too! It even sends all the [headers](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp#L288) i could possibly want to have :P (that sure makes it easy to do check on the http server end!)

jandrassy, 2020-08-02:
https://github.com/jandrassy/ArduinoOTA#esp8266-and-esp32-support

this ArduinoOTA library supports the esp boards only because the esp8266/esp32 OTA libraries don't support not native Ethernet. soon the esp8266 Arduino core will support Ethernet (W5000 series and enc28j60) as additional network interface with ESP8266WiFi library (now it has two net interfaces STA and SoftAP). the esp32 has Ethernet peripheral supported by the SDK, but can't work with W5000 and enc28j60.

closed 2020-08-02 by jandrassy

markg85, 2020-08-02:
Could you clarify that somewhere on the main page?
As the way it's currently described makes me believe it is supported.
Which is why i tried it.
And which is why i'm now spending a full day fixing my lights from the aftermath.. -_-

The problem is that, at first glance, it **seems** to work.
If the ESP8266 are not supported, perhaps just forcefully introduce a compiler error? some `#ifdef defined(ESP8266) force error` foo.

jandrassy, 2020-08-02:
OTA from IDE is supported and it was tested then. 
I never tested the OTASketchDownload example with ESP. It makes little sense to use it with esp, because the InternalStorageESP only wraps the esp Update object to be compatible with the OTA upload server of the library.

markg85, 2020-08-02:
It makes little sense if you know how it works. But that's **in hindsight**.
This stuff isn't easy to discover if you don't know the exact terms. In fact, you find ArduinoOTA first!

jandrassy, 2020-08-02:
the problem is platformio. in Arduino IDE you couldn't use this library with esp8266 without deleting the ArduinoOTA library bundled with the esp8266 boards package

markg85, 2020-08-02:
Please mention that on the main readme page.
I did in fact search for this library + platformio for a whole other reason, but wasn't able to find much in that combination.

Mention this on the readme is, in my opinion, quite important! I just spend about one and a half day of fixing stuff that wouldn't have been broken if this library would've just told me that it won't work. The rist, again, is that it "looks to be working" but doesn't fully.

jandrassy, 2020-08-02:
did you read the esp section of the readme?

markg85, 2020-08-02:
You really want me to go over it and provide feedback, it seems.
Fine.

`The ArduinoOTA library bundled with ESP8266 and ESP32 Arduino packages works only with native WiFi libraries.`

I interpreted this as my ESP8266 native wify libraries. So ESP8266WiFi.h which in turn gives me the WiFiClient class. Perhaps this was already the wrong assumption? But then i would really like to know what you mean by `native`. Here i assumed icould proceed.

`This library allows to upload a sketch to esp8266 or esp32 over Ethernet with Ethernet or UIPEthernet library. Upload over the native WiFi library works too.`

This too gives me the impression that i'm fine.

`To use this library instead of the bundled library, the bundled library must be removed from the boards package library folder. To override the configuration of OTA upload in platform.txt, copy the platform.local.txt file from extras folder of this library next to platform.txt file in boards package installation folder. Packages are located in ~/.arduino15/packages/ on Linux and %userprofile%\AppData\Local\Arduino15\packages\ on Windows (AppData is a hidden folder).`

This made no sense to me. I didn't seem to have the problem you mention (in hindsight apparently due to platform.io, i didn't know it back then). As i had no probably i just considered it to be not relevant.

`This library supports SPIFFS upload to esp8266 and esp32, but the IDE plugins have the network upload tool hardcoded to espota. It can't be changed in configuration. To upload SPIFFS, call the plugin in Tools menu and after it fails to upload over network, go to location of the created bin file and upload the file with arduinoOTA tool from command line. The location of the file is printed in the IDE console window. Upload command example (linux):`

Here it seemed to **again** mention that **it works** on ESP8266 and for the rest didn't seem to be relevant for my IDE so i ignored it.

So yes, i did read it. And it **very verry veeeeeeerrrrryyyyyy** clearly says that ESP8266 works!
I might be interpreting everything totally the opposite way.. But if i did, the readme isn't clear and left room for interpretation.
So for the third time, please fix it!

jandrassy, 2020-08-02:
when I wrote this there were no OTA **download** advanced examples. :-)

jandrassy, 2020-08-07:
this library was on my request removed from PlatformIO library register