JAndrassy/ArduinoOTA #43: Error flashing on Mega; OTA upload response: Payload size wrong

hanswesterv, 2020-08-28:
I use a board with a mega2560 and a esp8266ex. For the communication with the esp chip i use the WiFiEspAT library. Connecting to the wireless network is not a problem, this works nicely :). The next step is flashing the mega via the arduinoOta library.
Unfortunately this was not successful yet. When I use the following command: “C:\Users\Hans\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA -address 192.168.8.151 -port 65280 -username arduino -password password -sketch C:\Users\Hans\AppData\Local\Temp\arduino_build_392363/SerialWiFiOTA.ino.bin -upload /sketch -b”
I got the response:
“Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
Error flashing the sketch”

I placed some debug code (mentioned in another topic) to see what was causing the failure: “OTA upload response: Payload size wrong”. The contentLength = 16720, but read = 4163.
Do you have any idea how to solve this?

jandrassy, 2020-08-28:
what version of AT firmware? 1.7.4 or 2.1?
is the RSSI good?

btw, the upload tool should print the error too (from ArduinoOTA 1.0.4)

hanswesterv, 2020-08-28:
Thanks for the quick reaction.

I use 1.7.4 as AT firmware and arduinoOTA 1.0.5. the RSSI is -75.
I will try the 2.1 firmware.

jandrassy, 2020-08-28:
> Thanks for the quick reaction.
> 
> I use 1.7.4 as AT firmware and arduinoOTA 1.0.5. the RSSI is -75.
> I will try the 2.1 firmware.

no 1.7.4 is stable. 2.1 is too new.

hanswesterv, 2020-08-28:
Ok, I'll stay with version 1.7.4.
I placed the board close to the ap and now the RSSI is -43 but the result is the same.

hanswesterv, 2020-08-28:
Some more information:
 * The used board is like [https://robotdyn.com/mega-wifi-r3-atmega2560-esp8266-flash-32mb-usb-ttl-ch340g-micro-usb.html](https://robotdyn.com/mega-wifi-r3-atmega2560-esp8266-flash-32mb-usb-ttl-ch340g-micro-usb.html) 
 * I tried 115200 and 250000 as baud rate between the mega serial3 and the esp8266

jandrassy, 2020-08-29:
I tested it now to be sure and it works.
Do you have switches 5 and 6 OFF?

hanswesterv, 2020-08-29:
I did put both the dipswitchs to the off position, but i've some doubts about the quality of the dipswitch. I will replace the dipswitch and try again later this week

hanswesterv, 2020-09-02:
I tested the dipswitch, but is was ok. When trying some different things I saw that I probably have a problem with the esp8266. I re-flashed the module with the expressif firmware flash tool and I connected my laptop to the esp wifi network (got ip 192.168.4.2). According to windows the network connection is good (To be sure I connected an external antenna to the esp) but pinging to 192.168.4.1 is not that stable. Some pings are ok, but approx. half of them fail with a time out.

jandrassy, 2020-09-03:
the esp8266 has some RF calibration process done after some resets and stores the values to flash. when changing SDK versions (AT fw has a different version then esp8266 sketches), it is recommended to do a flash erase or flash blank.bin at the right address.

hanswesterv, 2020-09-04:
I tried to empty the erase the flash completely, but the effect is still the same. To rule out some things I ordered a new board. After flashing this new board the same way as the old one I got very different results. The new board works much better. After I flashed the optiboot bootloader from "my boards" I even got the "Sketch uploaded successfully" after uploading a new sketch via the arduinoOTO command line tooling. Unfortunately the new sketch is not running after a flash or reset. The old one is still active... :S. 

Because the new board with the same software /settings works way better than the old board I think that my original issue was some kind of a hardware issue.

hanswesterv, 2020-09-05:
After doing a clean install of all arduino stuff, also the flashing went OK. I would like to thank you for your time / help.

closed 2020-09-05 by hanswesterv