JAndrassy/ArduinoOTA #44: Arduino Mega 2560 - boards needs hardware reset after every OTA flash

peros550, 2020-08-30:
Thank you for this nice library. 

I'm facing an issue and I'm not sure how to troubleshoot. 

Every time I try to upload new code using OTA the board does not run the new code unless I hardware reset it first. 

The whole time, the board is responsive in ping tests. 

Similarly, once I upload the new code over OTA, I cannot upload again unless I hardware reset it first. 

Is this behavior expected?

jandrassy, 2020-08-30:
it is not expected. that would not be  very useful.
after the reset the updated sketch runs?

peros550, 2020-08-30:
yes runs fine. After reset I can use again OTA. How can I troubleshoot?

jandrassy, 2020-08-30:
> yes runs fine. After reset I can use again OTA. How can I troubleshoot?

but is  there the new version of the sketch?

I suspect you didn't change the bootloader to Optiboot with the copy_flash_pages function.

peros550, 2020-08-30:
I am pretty sure I changed the bootloader but I cant't remember details. I had to wire an arduino UNO to the Arduino Mega and used the UNO as a programmer. 

When I upload a sketch, the IDE says all good, but the board freezes. Then I reset the board and the new sketch runs fine.

jandrassy, 2020-08-30:
> I am pretty sure I changed the bootloader but I cant't remember details. I had to wire an arduino UNO to the Arduino Mega and used the UNO as a programmer.
> 
> When I upload a sketch, the IDE says all good, but the board freezes. Then I reset the board and the new sketch runs fine.

but if you upload a different sketch, which sketch runs after the reset? try to upload Blink

peros550, 2020-08-30:
I am 100% sure that when I OTA upload a new sketch , the updated one run after reboot. I have doubled confirmed that. 

When I upload via USB I use "Arduino Mega 2560 (Optiboot)". In fact, when I also upload through OTA I use the same option.

jandrassy, 2020-08-31:
> I am 100% sure that when I OTA upload a new sketch , the updated one run after reboot. I have doubled confirmed that.
> 
> When I upload via USB I use "Arduino Mega 2560 (Optiboot)". In fact, when I also upload through OTA I use the same option.

then it looks like you do everything right.

the bootloader should do the reset using watchdog reset. I don't know what could e wrong with that. maybe try to burn the bootloader again.

peros550, 2020-08-31:
I have two 2560 boards with the above issue. Today, I followed your advise and changed bootloader on a third one. This board seems to work fine. I will try reflashing the other two and see how it goes. 

Thank you for your support!

closed 2020-08-31 by peros550

BartArIen, 2020-10-29:
Hi Peros550, 

Did you still have the problem with the 2 others boards because I have the same issue and reburned the optiboot but no succes.

jandrassy, 2020-10-29:
> Hi Peros550,
> 
> Did you still have the problem with the 2 others boards because I have the same issue and reburned the optiboot but no succes.

do you have the Optiboot with the copy_flash_pages function?

peros550, 2020-10-29:
hi guys, unfortunately I still have this issue. I never came back because I was getting contradictive results in my tests, hence not entirely sure the problem is with this library or my code or the combination of both. To be more specific, If I run any of the examples that come with the library, OTA functions works always. When I add my code and use it on a Mega2560 which is connected to relays, buttons, then OTA function works only for the first time.

jandrassy, 2020-10-30:
> hi guys, unfortunately I still have this issue. I never came back because I was getting contradictive results in my tests, hence not entirely sure the problem is with this library or my code or the [...]

powering?

Fristi10, 2020-10-30:
The last 4 hours I did some random upload tests with the same sketchfile and +- 5 times it succeeded and then it broke down. Even when I uploaded the same sketch file through usb the Arduino IDE didn't show the ip address but the program was working ok because I was collecting some network data.  I use a atmega2560 with a w5100 shield. Powering should be okey because i used the same setup for like 8 month without problems and the programs runs fine so..

maybe the Arduino IDE is the problem but it is up to date and i restarted it multiple times.

jandrassy, 2020-10-31:
maybe the problem is that the watchdog reset used for the self reset of the MCU doesn't reset the W5100. the Ethernet library does a software reset of the W5100, but it may be not enough

peros550, 2020-10-31:
@jandrassy is the ArduinoOTA relying on watchdog timer to do the reset after OTA flash? If yes, then maybe this is reason why my code in combination with ArduinoOTA cannot OTA flash for the second time? (It can flash for a first time) 

To keep it simple, here is my code simplified to showcase how I use the watchdog timer. 

I am also using mqtt.h library. I had to use a watchdog timer because every few days the code used to stuck and couldn't do anything but hardware reset to make it work again.  Hence, with the watchdog timer, although not an elegant solution it now works. 

```
void setup() {
wdt_disable();

//some pin initialization

//Serial port initialization
Serial.begin(115200);
while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
}
Serial.print("start ....");

//Ethernet initialization
Ethernet.begin(mac,ip, myDns, gateway, subnet);
delay(1500);
lastMQTTReconnectAttempt = 0;
Serial.print("server is at ''");
Serial.println(Ethernet.localIP());
//**********

//Watchdog start
watchdogOn();

//mqtt initialization
ethernet_UDP.begin(localPort);
Serial.println("Ethernet UDP Start....");

client.begin(mqtt_server, ethClient);
client.onMessage(messageReceived);
client.setWill(willTopic, "Offline", true,2);
connect();

//ArduinoOTA initialization
ArduinoOTA.begin(Ethernet.localIP(), "ArduinoEisodos1", "password", InternalStorage);
  
}

void loop(){

wdt_reset();
      
// check for updates
ArduinoOTA.poll();

 wdt_reset();
 client.loop();

//some other code
wdt_reset();

}

void watchdogOn() {
 
// Clear the reset flag, the WDRF bit (bit 3) of MCUSR.
MCUSR = MCUSR & B11110111;
 
// Set the WDCE bit (bit 4) and the WDE bit (bit 3) 
// of WDTCSR. The WDCE bit must be set in order to 
// change WDE or the watchdog prescalers. Setting the 
// WDCE bit will allow updtaes to the prescalers and 
// WDE for 4 clock cycles then it will be reset by 
// hardware.
WDTCSR = WDTCSR | B00011000; 

// Set the watchdog timeout prescaler value to 1024 K 
// which will yeild a time-out interval of about 8.0 s.
WDTCSR = B00100001;

// Enable the watchdog timer interupt.
WDTCSR = WDTCSR | B01000000;
MCUSR = MCUSR & B11110111;

}
```

jandrassy, 2020-10-31:
ArduinoOTA doesn't call wdt_reset while writing uploaded binary to flash and it takes some seconds, so turn-off watchdog or set longer time before the upload
in setup()
```
ArduinoOTA.beforeApply(otaStart);
```
```
void otaStart() {
  // reconfigure watchdog here
}
```
you can configure it back in onError

peros550, 2020-10-31:
@jandrassy thank you! Really appreciate your help! That must be it.

peros550, 2020-10-31:
@jandrassy unfortunately, the above didn't help. In fact, I removed all references of watchdog timer and tried to OTA flash.  OTA flash completes, but then the board gets stuck and needs hardware reset. The problem seems to be related with the reset. 

I remember, when I wanted to start using the watchdog timer, to solve another problem, I had faced difficulties using the timer with the common code patterns I found on the web. Then, I came across a [forum page](https://forum.arduino.cc/index.php?topic=176329.0), in which people were reporting issues with the watchdog timer on Mega2560 boards. 

The above posted function (_watchdogOn()_) that is used to enable watchdog specifically for Mega2560 seemed to solve the problem of watchdog for my board.

Ofcourse I understand it's very difficult to solve a problem which you cannot reproduce. 

Any ideas or further help would be greatly appreciated.

jandrassy, 2020-10-31:
the watchdog reset after OTA upload is done in bootloader with a function used in many places in Optiboot

jadero, 2025-08-14:
Has there been any progress or a solution to the problem?