JAndrassy/ArduinoOTA #46: Add MKR WiFi 1010 to "Boards tested" list

kpfleming, 2020-10-22:
I have this working now using InternalStorage. When using SDStorage the upload completed successfully and the Arduino rebooted, but SDU didn't apply the update to the sketch. Since I have plenty of Flash memory space for my sketch, I opted not to try to debug that any further :-)

jandrassy, 2020-10-22:
https://github.com/arduino/ArduinoCore-samd/issues/563

quentin-gilli, 2020-11-15:
Hi Fleming, 

Everything works with your MKR? 
I can't get mine to work, while I have no problem with my ESP32.
Are there any differences that can explain this ?

jandrassy, 2020-11-15:
> Are there any differences that can explain this ?
a completely  different library maybe?

I assume with ESP32 you use the library bundled with the ESP32 boards package, while with MKR you try this library.
What is the problem?

closed 2020-12-13 by jandrassy