JAndrassy/ArduinoOTA #48: MKR 1010 what is wrong ?

quentin-gilli, 2020-11-15:
Hi, 

Sorry to disturb, but after a few hours of searching I couldn't find an answer to my concern. 

I uploaded the **WiFi101_OTA** sketch on my **MKR 1010**, and let the card connect to WiFi.
And no port is showing, so I try : 
- Bonjour browser on my Mac --> Nothing appear
- Bonjour browser on my Android --> Nothing appear
- OTA Upload from IDE without 'network port' --> Error "Connecting to board ... Error flashing the sketch"
- Spend a lot of time trying to change my OS / Network parameters...
- Finally I uploaded the **BasicOTA sketch on my Esp32 and everything work fine** !

So why my **MKR 1010** won't show itself :( ?

Thanks you for your help :)

jandrassy, 2020-11-15:
what version of SAMD boards package do you use?
does the WiFiWebServer example work?
can you `ping` the board?

quentin-gilli, 2020-11-15:
Hi, 

Thanks you for your time, 
My SAMD package version is 1.8.9
And my WiFiNINA firmware onboard is 1.3.0 (not the latest, but I cannot install the 1.4.1 through the IDE uploader)
Yes the WiFiWebServer example words and also the pings : 
[image]

jandrassy, 2020-11-15:
does `ping` work with WiFi101_OTA example uploaded?

quentin-gilli, 2020-11-15:
oh right sorry, 
yes ping works with WiFi101_OTA example uploaded. 
So I double check the programmers.txt and correct a mistake in the IP adresse :( 
Now, the OTA Upload from IDE without 'network port work fine !
Of course still no network port showing in IDE, unlike with the ESP32, but the main goal is reached !

closed 2020-12-09 by jandrassy