JAndrassy/ArduinoOTA #49: Support NANO 33 BLE via Bluetooth?

tangtangdn, 2020-11-18:
Hi,

Thank you very much for your library. I see you said this library supports "Nano 33 BLE OTA over TCP should work too (I have nRF5 support in the library)". I am using Nano 33 BLE which has no WiFi function. Could you kindly give an example of how to update the firmware of NANO 33 BLE with ArduinoOTA via OTA? Thank you very much.

Kind regards,
Tang

jandrassy, 2020-11-19:
Nano 33 BLE is not supported yet. the code for nRF51 doesn't work for nRF52840

closed 2020-11-19 by jandrassy

tangtangdn, 2020-11-19:
> Nano 33 BLE is not supported yet. the code for nRF51 doesn't work for nRF52840

Thank you very much for the prompt response and important information.

tangtangdn, 2020-11-19:
Dear Juraj,

Thank you so much for your prompt response. I see your library supports WiFiNINA - MKR 1010, MKR 4000, Nano 33 IoT and any supported MCU with attached ESP32 as SPI network adapter with WiFiNINA firmware. Is it can be used to upload the program (sketch) to Nano 33 IoT through wireless method (WiFi or Bluetooth)? May I just try this WiFi101_OTA example<https://github.com/jandrassy/ArduinoOTA/tree/master/examples/WiFi101_OTA>?
Many thanks in advance and have a great day.

Kind regards,
Tang

jandrassy, 2020-11-19:
SAMD21 was already supported with the WIFi101_OTA library. (ArduinoOTA is enhanced version of it)
WiFiNINA library is a supported way of upload.
ArduinoOTA library (and WIFi101_OTA library) supports only TCP upload. Alternatively and it can be used with other transport way to store and apply the update.

tangtangdn, 2020-11-19:
Dear Juraj,
Thank you so much for your prompt and professional response.
Kind regards,
Tang

andreacab2, 2022-03-10:
Is it supported now by any chance?

jandrassy, 2022-03-10:
no. Nano BLE is not supported ad BLE is not supported in this. but I think nRF52 SDK supports BLE OTA

baconcheese113, 2022-08-12:
I'm currently trying to switch from the nano 33 IoT to the Seeed XIAO nRF52840 (same processor as nano 33 BLE)

I had only been using [InternalStorage.cpp](https://github.com/JAndrassy/ArduinoOTA/blob/master/src/InternalStorage.cpp) to `write()` and `apply()` the update. Just to clarify, are you saying that the `ARDUINO_ARCH_NRF5` code will not work for the nRF52 to write to the flash?

JAndrassy, 2022-08-12:
nRF51 and nRF52 boards  are in ArduinoOTA only supported with the sandeepmistry/arduino-nRF5 core. this core doesn't support nRF52840  (because Nordic SDK version which supports nRF52840  has support for nRF51 removed).