JAndrassy/ArduinoOTA #53: Support for SAMD51

jasonthenderson, 2020-12-24:
I am trying to add OTA update via wifi download from web to Flash for a SAMD51J20, specifically the Titano (https://learn.adafruit.com/adafruit-pyportal-titano).  Any pointers on how to do this would be appreciated.  

I am pretty basic in my skillset.

jandrassy, 2020-12-24:
see if you can enhance InternalStorage.cpp

jandrassy, 2021-07-30:
@jasonthenderson can you please try the SAMD51 support

doboszm, 2021-08-03:
@jandrassy I just tested it with an adafruit matrixportal using adafruit's wifinina library and a custom board that uses the matrixportal bootloader

jandrassy, 2021-08-03:
> @jandrassy I just tested it with an adafruit matrixportal using adafruit's wifinina library and a custom board that uses the matrixportal bootloader

MCU. did it work?
btw the bootloader is not important for ARM

doboszm, 2021-08-03:
> > @jandrassy I just tested it with an adafruit matrixportal using adafruit's wifinina library and a custom board that uses the matrixportal bootloader
> 
> MCU. did it work?
> btw the bootloader used is important only for AVR

ATSAMD51J19A, worked for both serial and network download with a slight modification to both examples (using Adafruit wifinina library instead of Arduino's)

PR #104 (2021-07-30): Initial SAMD51 support
Tested with Seeed Wio Terminal

Closes #53
closed 2021-08-03 by jandrassy