JAndrassy/ArduinoOTA #55: Is it possible to use this library with VS Code?

izamoramx, 2021-01-04:
Hello @jandrassy ,

Is there any documentation on how to configure VS Code projects to be able to upload via IP address with your library? Or could you share in here how to configure the VS Code to make it work.

I have everything working on Arduino IDE but I would like to be able to do the same on VS Code. (all the code compiles, just fails during upload).

Thanks,

jandrassy, 2021-01-04:
Platformio?

izamoramx, 2021-01-04:
> Platformio?

No. Just VS Code with the MS Arduino for Visual Studio Code plug in.

jandrassy, 2021-01-04:
you can use the library, but I know nothing about the VS Arduino plugin. does it use Arduino boards packages? will it read platform.local.txt for additional build and upload commands? does it show the network port?
if not, you can do upload from command line. the upload command is in platform.local.txt in extras folder. .bin file must be uploaded so for AVR the hex file must first be converted to .bin

izamoramx, 2021-01-04:
I haven't been able to find anything specific saying that it does or doesn't. As far as I am aware, from the available information, it is just an IDE for Arduino. I requires arduino IDE to be installed and it uses it to compile. It uses all the preinstall libraries. Other than the specific features of VS Code, I don't see it using it's own libraries or code to generate the binaries so I believe it just interfaces with Arduino IDE. Here is the link to the project FYI:

https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino

I believe it works the same as Eclipse, and I think I saw you mentioned on a conversation that you use Eclipse to develop for Arduino. Is there any configuration required to make OTA work on Eclipse? maybe the same thing could solve the issue on VS Code.

jandrassy, 2021-01-05:
I use Eclipse Sloeber plugin and that doesn't require Arduino IDE installation but uses Arduino boards packages (has own way to install them). But Sloeber doesn't read platform.local.txt patch so the changes must be merged to platform.txt or as I use it with the [my_boards](https://github.com/jandrassy/my_boards) concept.

anklimov, 2021-01-05:
It's absolutely possible to use lib with platformio+vsc, target config example from platformio.ini below:
Note: be sure you have arduinoOTA utility somewhere in PATH
`
[env:esp32-wifi]
platform = espressif32
framework = arduino
monitor_baud = 115200
board = esp32-evb
upload_command = arduinoOTA -address 192.168.1.1 -port 65280 -username arduino -password password -b -upload /sketch -sketch $SOURCE
upload_protocol = custom 
`

izamoramx, 2021-01-12:
> It's absolutely possible to use lib with platformio+vsc, target config example from platformio.ini below:
> Note: be sure you have arduinoOTA utility somewhere in PATH
>
> [env:esp32-wifi]
> platform = [...]

@anklimov  thanks for the reply. Do you know if there is a similar configuration for just using the VS Code Arduino plugin instead of using PlatformIO?

closed 2021-01-29 by jandrassy