JAndrassy/ArduinoOTA #56: OTA is not working with long ESP32 binaries (about 1M)

anklimov, 2021-01-05:
The maximum firmware length depended from Wifi link speed. In most cases, OTA failed if binary about 800-900K

jandrassy, 2021-01-06:
try larger buffer here
https://github.com/jandrassy/ArduinoOTA/blob/3ef07acd8a2697bfcdd92fe58db4a3e555dd79a2/src/WiFiOTA.cpp#L292

anklimov, 2021-01-08:
Tried 4096 bytes 
There is no significant progress with speedup:

[E][Updater.cpp:272] end(): premature end: res:0, pos:671525/1060288

jandrassy, 2021-01-08:
4096 is too much. try 512 or 1024 bytes

anklimov, 2021-01-08:
Same behaviour with 1024

[E][Updater.cpp:272] end(): premature end: res:0, pos:622373/1060288

it looks as timeout issue for http POST operation

anklimov, 2021-06-11:
Hi! is some idea how to fix timeout? May you point me on arduinoOTA upload utility sources? Possible, easiest way to increase timeout on client side, most probably, this utility designed for quite small sketches initially

jandrassy, 2021-06-11:
> Hi! is some idea how to fix timeout? May you point me on arduinoOTA upload utility sources? Possible, easiest way to increase timeout on client side, most probably, this utility designed for quite small sketches initially

https://github.com/arduino/arduinoOTA

anklimov, 2021-06-11:
Ok, Thanks
It's really helps !	

main.go:
`var httpClient = &http.Client{
		Timeout: time.Second * 10,
`
After changing timeout to 30 - OTA working perfectly with > 1M firmware

jandrassy, 2021-06-12:
btw, for ESP32 it is better to use a PHY layer LAN module like LAN8720 supported by the ESP32 MAC peripheral and the Arduino boards support package. then you can use the standard esp32 networking library (WiFi.h even it is Ethernet) and the standard ESP32 ArduinoOTA library

closed 2021-07-13 by jandrassy