JAndrassy/ArduinoOTA #57: Compatibility With Arduino Uno Wifi Rev 2

bryce13950, 2021-01-22:
When compiling for the Arduino Uno Wifi Rev 2 I am receiving the following error...

```
Arduino: 1.8.13 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

/user-path/Documents/Arduino/libraries/ArduinoOTA/src/ArduinoOTA.h:133:2: warning: #warning "Network library not included or not supported" [-Wcpp]
 #warning "Network library not included or not supported"
  ^~~~~~~
In file included from /user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:16:0:
/user-path/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/boot.h:116:6: error: #error AVR processor does not provide bootloader support!
 #    error AVR processor does not provide bootloader support!
      ^~~~~
/user-path/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/boot.h:127:4: error: #error Cannot find SPM Enable bit definition!
 #  error Cannot find SPM Enable bit definition!
    ^~~~~
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp: In constructor 'OTAStorage::OTAStorage()':
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:31:19: error: 'SPM_PAGESIZE' was not declared in this scope
         PAGE_SIZE(SPM_PAGESIZE),
                   ^~~~~~~~~~~~
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:31:19: note: suggested alternative: 'E2PAGESIZE'
         PAGE_SIZE(SPM_PAGESIZE),
                   ^~~~~~~~~~~~
                   E2PAGESIZE
In file included from /user-path/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/io.h:99:0,
                 from /user-path/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/avr/include/avr/pgmspace.h:90,
                 from /user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/String.h:31,
                 from /user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/Print.h:24,
                 from /user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/Stream.h:25,
                 from /user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/Client.h:22,
                 from /user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/ArduinoAPI.h:29,
                 from /user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/Arduino.h:23,
                 from /user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.h:27,
                 from /user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:1:
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:42:22: error: '__SPM_REG' was not declared in this scope
   uint8_t highBits = boot_lock_fuse_bits_get(GET_HIGH_FUSE_BITS);
                      ^
In file included from /user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:16:0:
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:42:22: error: '__BOOT_LOCK_BITS_SET' was not declared in this scope
   uint8_t highBits = boot_lock_fuse_bits_get(GET_HIGH_FUSE_BITS);
                      ^
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:42:22: note: suggested alternative: '__LOCK_BITS_EXIST'
In file included from /user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/HardwareSPI.h:21:0,
                 from /user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/ArduinoAPI.h:31,
                 from /user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/Arduino.h:23,
                 from /user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.h:27,
                 from /user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:1:
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:44:24: error: 'FUSE_BOOTRST' was not declared in this scope
   if (!(highBits & bit(FUSE_BOOTRST))) {
                        ^
/user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/Common.h:70:25: note: in definition of macro 'bit'
 #define bit(b) (1UL << (b))
                         ^
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:44:24: note: suggested alternative: 'FUSE_BOOTEND'
   if (!(highBits & bit(FUSE_BOOTRST))) {
                        ^
/user-path/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/Common.h:70:25: note: in definition of macro 'bit'
 #define bit(b) (1UL << (b))
                         ^
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:45:29: error: 'FUSE_BOOTSZ1' was not declared in this scope
     byte v = (highBits & ((~FUSE_BOOTSZ1 ) + (~FUSE_BOOTSZ0 )));
                             ^~~~~~~~~~~~
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:45:29: note: suggested alternative: 'FUSE_BOOTEND'
     byte v = (highBits & ((~FUSE_BOOTSZ1 ) + (~FUSE_BOOTSZ0 )));
                             ^~~~~~~~~~~~
                             FUSE_BOOTEND
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:45:48: error: 'FUSE_BOOTSZ0' was not declared in this scope
     byte v = (highBits & ((~FUSE_BOOTSZ1 ) + (~FUSE_BOOTSZ0 )));
                                                ^~~~~~~~~~~~
/user-path/Documents/Arduino/libraries/ArduinoOTA/src/OTAStorage.cpp:45:48: note: suggested alternative: 'FUSE_BOOTEND'
     byte v = (highBits & ((~FUSE_BOOTSZ1 ) + (~FUSE_BOOTSZ0 )));
                                                ^~~~~~~~~~~~
                                                FUSE_BOOTEND
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

A lot of this seems like it could be a matter of changing names, but I am not entirely sure how in-depth it would be to get it to work on that board. Has there been any work into getting it to work, and could anyone with expertise in this library help guide me on what I might need to do to get it to compile?

jandrassy, 2021-01-23:
1. new mega series AVRs are not supported. they have a different memory controller then classic AVR MCUs
2. it has less then 64kB flash

https://github.com/jandrassy/ArduinoOTA#supported-micro-controllers

closed 2021-01-23 by jandrassy

bryce13950, 2021-01-23:
Is it completely impossible to make it happen? It has 48kb, which is obviously less, but it’s not substantially less. The program I am trying to flash is going to be 16k, so that should still be enough. I must clarify as well that this issue is not a support issue, but a question on what I would need to change in the library to add compatibility.

jandrassy, 2021-01-23:
you can try it

bryce13950, 2021-01-24:
Any idea on where I need to start in the code?

jandrassy, 2021-01-24:
now there are two implementations of InternalStorage. one for ARM MCU which can write directly to flash memory. for classic AVR there is a separate implementation. I guess the new AVR MCUs like the 4809 can directly write to flash memory so the implementation could go into InternalStorage with ifdefs.
other option is to write a SD bootloader to apply the update over the SD card. I guess it could be made based on existing bootloader for 4809 and one of the SD bootloaders I mention in README.

jandrassy, 2021-01-26:
sorry the ARM way of InternalStorage is not right. it requires a function running in RAM and AVR can't do that

with 48kB flash and 1kB bootloader the sketch size would be limited to 18 kB. an ArduinoOTA example is 13 to 15kB because of the size of networking libraries. that would leave only 3kB flash for the project's code

the SD bootloader way is feasible

bryce13950, 2021-02-17:
Could you reopen this, so that I can find it quicker when I am working through this?

bryce13950, 2021-02-17:
I have just started working through getting this to compile for the SD Bootloader route, and I am running into an issue where the compiler says that `SPM_PAGESIZE` is not defined among other things. I have found that this should be defined in avr/io.h, which seems to be defining various values based on the board. Now with the 4809, this value is not defined. I have found some articles that say what the value should be, and I am contemplating simply defining the values if the board is 4809 and if the value is not previously defined. Can you for see any problems with this approach that I may not be thinking of?

jandrassy, 2021-02-17:
> I have just started working through getting this to compile for the SD Bootloader route, and I am running into an issue where the compiler says that `SPM_PAGESIZE` is not defined among other things. 

in bootloader? how does it the Uno WiFi bootloader?

bryce13950, 2021-02-19:
Could you rephrase your question? I don’t quite understand.

jandrassy, 2021-02-19:
> the compiler says that `SPM_PAGESIZE` is not defined among other things. 

where? copy the error messages here

bryce13950, 2021-02-22:
Thanks a ton for your help on this! I really appreciate it. I will give the SD card help, and if I have any issues that seem to be caused by the library I will open a different issue. If I get it to work properly, would you like me to update the readme to reflect that?

jandrassy, 2021-02-22:
> Thanks a ton for your help on this! I really appreciate it. I will give the SD card help, and if I have any issues that seem to be caused by the library I will open a different issue. If I get it to work properly, would you like me to update the readme to reflect that?

README has "any board with MCU with SD bootloader" and WiFiNINA is in the list of supported libraries..

SpyGuy0215, 2022-04-17:
Any update on this? I would love this feature, but unfortunately I don't have the technical skills to develop something like this :/