JAndrassy/ArduinoOTA #58: Is possible to change the OTA port?

iikiik77, 2021-01-28:
I really impressed with testing this wonderful library. It's not easy at first time, It works well in ESP32.
Is possible to change the OTA port in FW?
I just have seen OTA_PORT in ArduinOTA.h
const uint16_t OTA_PORT = 65280;

template <class NetServer, class NetClient>
class ArduinoOTAClass : public WiFiOTAClass {

private:
  NetServer server;

public:
  ArduinoOTAClass() : server(OTA_PORT) {};

I would like to set different port each device like below.
ArduinoOTA.setPort(1070);

Could you give me a help?
Thanks so much.

jandrassy, 2021-01-28:
why do you need a different port? "different port (for) each device". what for? how would you change port for upload tool?

iikiik77, 2021-01-29:
My device is connected with a router which has a several device. 
ex)
External IP(220,140,22,70) - router - internal IP(192.168.0,1) - first device             
                                                          internal IP(192.168.0,2) - second device
                                                          internal IP(192.168.0,3) - third device
                                                          internal IP(192.168.0,3) - fourth device
I would like to try update each device by using a port forward of router.
I can only update one device and I should have to change a port forward setting of router for next device.
curl --request POST --data-binary @"C:\firmware.bin" http://arduino:password@220,140,22,70:65280/sketch
                                             external port          internal port
External IP(220,140,22,70) -   65280-  router   -   65280-  internal IP(192.168.0,1) - first device 
===>
If I could change the OTA port.
                                       external port          internal port
External IP(220,140,22,70) -  100-       router -         100          internal IP(192.168.0,1) - first device             
                                              101-                            101         internal IP(192.168.0,2) - second device
                                              102-                            102        internal IP(192.168.0,3) - third device
                                              103-                            103       internal IP(192.168.0,3) - fourth device
I thank that I can access each device by using different port in remote site. This is what i want.
like below 
curl --request POST --data-binary @"C:\firmware.bin" http://arduino:password@220,140,22,70:100/sketch
curl --request POST --data-binary @"C:\firmware.bin" http://arduino:password@220,140,22,70:101/sketch
curl --request POST --data-binary @"C:\firmware.bin" http://arduino:password@220,140,22,70:102/sketch
curl --request POST --data-binary @"C:\firmware.bin" http://arduino:password@220,140,22,70:103/sketch   
                              
Actually I am not family with OTA system now.
It very very fun for me your OTA. My system is ESP32 and W5500(Ethernet) which working well now. It looks wonderful.

jandrassy, 2021-01-29:
is it not possible to forward to a different port?

iikiik77, 2021-01-29:
Dear jandrassy

I did it.
I have changed OTA_PORT of ArduinoOTA by connecting extern variable.
If I have set OTA_PORT in FW, it update extern OTA_PORT of ArduinoOTA.

Thanks so much for your care and your very great works.

jandrassy, 2021-01-29:
but you could set in router port forwarding
External IP(220,140,22,70) - **65281**- router - 65280- internal IP(192.168.0,1) - first device

closed 2021-01-29 by jandrassy