JAndrassy/ArduinoOTA #71: Atmega2560 seems to hang after upload

burnt-past, 2021-03-08:
After the MCU has copied to flash and a reset is triggered, the MCU gets stuck in an unresponsive state. Power cycling seams to complete the update, though this seems kind of counter intuitive.

I am assuming that the watchdog timer is not getting disabled in the bootloader by default, causing the MCU to get caught in a boot loop. I would look further into this, but I can't find the source code for the bootloader that you built.

Any thoughts on this?

jandrassy, 2021-03-09:
the source of the bootloader is in MCUDude's cores or in the official Optiboot repo with my unmerged PR

jandrassy, 2021-04-25:
what networking shield/module do you use?

peros550, 2021-04-25:
I'm having this issue using Arduino Ethernet Shield V1

jandrassy, 2021-04-26:
> I'm having this issue using Arduino Ethernet Shield V1

did you debug to determine if the sketch doesn't start at all or it hangs in setup for example in Ethernet.begin()?

peros550, 2021-04-26:
Thanks for your response. I have the same issue in two 2560 boards. I use these to control external lights of my house. The two slightly different sketches run fine for months. The moment I upload an updated sketch I get this strange behavior as OP mentioned. The uploaded sketch seems to be in the board but it won't run until I power cycle the board.

jandrassy, 2021-04-26:
I can't debug your problem on your hardware. You must do it. Connect the Mega to computer add debug prints to sketch, open Serial Monitor in IDE, do OTA upload from a second instance of the IDE and see where it hangs after reset.

burnt-past, 2021-04-26:
I am not using a networking module. I actually have all the network related code commented out. I am using a USB chip to transfer the data locally. After I trigger the reset through the library, none of my initializing code runs. After a power cycle, it seems that the update has completed.

jandrassy, 2021-04-27:
@burnt-past, so you use code like the OTASketchDownload examples, but you read the binary over USB? why?

burnt-past, 2021-04-27:
For the particular project I didn't need any ethernet connectivity.

jandrassy, 2021-04-27:
> 
> 
> For the particular project I didn't need any ethernet connectivity.

I meant why use self-update instead of use normal avcrdude upload over USB and bootloader?

burnt-past, 2021-04-27:
> I meant why use self-update instead of normal avcrdude upload over USB and bootloader?

Oh, sorry for the confusion. I meant I use a usb stick with a CH376s chip. My bad on responding when I was in the middle of something.

burnt-past, 2021-09-06:
So, I found that the watchdog timer was putting the chip in a boot loop. Doing a jump to the bootloader start address instead of using the libraries reset function seemed to fix the issue.

I'm using the largest boot size, so this worked for me.
`fptr_t resetFunc = (fptr_t)0x1F000;`

Then to call it just use:
`resetFunc();`

jandrassy, 2021-09-06:
> So, I found that the watchdog timer was putting the chip in a boot loop. Doing a jump to the bootloader start address instead of using the libraries reset function seemed to fix the issue.

Optiboot turns off the Watchdog so boot loop doesn't happen

closed 2021-09-06 by jandrassy