JAndrassy/ArduinoOTA #74: OTA stopped working

SteveRMann, 2021-03-26:
I have a project on a Wemos D1 Mini that has been running for a year or so with occasional updates by OTA.

This morning I did another update- or I tried- but OTA fails with:
[ERROR]: No response from device

I spent the morning trying to identify if anything in my sketch could have caused this, but nothing looked suspicious.  I tried reducing my program to a simple test, but still no OTA.  Next, I tried running the BasicOTA example sketch. Again, no connection. I tried on a new Wemos D1 Mini and a new NodeMCU board.  No connection.

The device, by the way, receives and displays data over MQTT, so its WiFi is working. Except for OTA

The only thing that has changed on my network is that I replaced my router two weeks ago. The old router did not do WPA and I couldn't use the latest Tasmota or ESPHome.

Does anyone have any ideas why my OTA stopped working, or where to look for a solution?

jandrassy, 2021-03-26:
you don't use this library. you use the  ArduinoOTA library bundled with the esp8266 arduino boards support package

closed 2021-03-26 by jandrassy

SteveRMann, 2021-03-26:
Who manages that library?

jandrassy, 2021-03-27:
> Who manages that library?

the esp8266 Arduino core community

SteveRMann, 2021-03-29:
SOLVED
Thanks for the feedback. The problem, it turns out, is that my PC Ethernet type had been changed to "Public" from "Private".  Resetting the type to "Private" fixed the problem.