JAndrassy/ArduinoOTA #84: ESP32 no chance to OTE update

jhnwmr, 2021-05-04:
Hello, I can successfully flashed my ESP with the ArduinoOTA library. I am using a W5500 SPI Ethernet module on the ESP with the Ethernet2 library.
I have tested the OTEthernet example and can also successfully connect to my network. 
However, the ESP32 **does not show up** in the Arduino IDE Ports menu for the OTA(E) update.
What am I doing wrong?
Here is the example code:
```
#include <SPI.h>
#include <Ethernet2.h>
#include <ArduinoOTA.h>

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

void setup() {
  Serial.begin(115200);
  while (!Serial);
  Ethernet.ini(27);
  Serial.println("Initialize Ethernet with DHCP:");
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
  } else {
    Serial.print("  DHCP assigned IP ");
    Serial.println(Ethernet.localIP());
  }
  ArduinoOTA.begin(Ethernet.localIP(), "Arduino", "password", InternalStorage);
}

void loop() {
  ArduinoOTA.poll();

  // add your normal loop code below ...
}
```

An here is the debug output:
```
Initialize Ethernet with DHCP:
  DHCP assigned IP 192.168.170.180
```

jandrassy, 2021-05-04:
Ethernet2 is not supported

jhnwmr, 2021-05-05:
I have also tried the normal Ethernet library, but it always comes up: `Failed to configure Ethernet using DHCP`
I use the HSPI interface of the ESP32, which I changed in the Ethernet2 library. With the normal Ethernet Library I have worked with `SPIClass spiETH(HSPI);` and `spiETH.begin(14, 12, 13, 27);`, but this has unfortunately always brought an error.
Do you have any ideas?

jandrassy, 2021-05-05:
do you have the network port in IDE for normal Esp32 arduino OTA over WiFi?

jhnwmr, 2021-05-05:
No via WLAN is unfortunately not an option, that's why I came to your library with the OTE update. 
The problem is only the original Ethernet library, which does not support the W5500.

Where is the problem with the Ethernet2 library, which is why I can not use your library.

jandrassy, 2021-05-05:
Ethernet library supports W5100, W5200 and W550 since 2018

jhnwmr, 2021-05-05:
Okay I have tested it again now and I can connect with the Ethernet Library. However, I can not make an update. It always comes the following error message:
```
python /Users/Service/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc2/tools/espota.py -i 192.168.170.180 -p 65280 --auth= -f /var/folders/0x/l_p_wffx7876z_6d9wg3n6800000gn/T/arduino_build_999140/OTEthernet.ino.bin 
Sending invitation to 192.168.170.180 ..........
14:49:29 [ERROR]: No response from the ESP
```

jandrassy, 2021-05-06:
because it doesn't use espota
https://github.com/jandrassy/ArduinoOTA#esp8266-and-esp32-support

jhnwmr, 2021-05-06:
Okay thanks, now I can do OTE Updates!

closed 2021-05-06 by jhnwmr