JAndrassy/ArduinoOTA #91: Sketch Size Limitation / Overlap

burnt-past, 2021-06-02:
I'm not sure how accurate this is, but if an already loaded firmware is larger than **_50%_** on an Atmega2560, the upload seems to corrupt the device.

I am using a small preliminary firmware that is only **_20000U_**. That gets loaded in before the larger update. This firmware just contains the same code used to upload the preliminary firmware. The idea was I could get past the **_50%_** firmware limit by uploading a small one that just loads the larger one.

If the currently loaded firmware is already past that halfway mark, it seems to not throw any errors from the library. Although after a reboot is triggered the device ends up corrupt.

jandrassy, 2021-06-02:
InternalStorage stores the uploaded binary in the upper half of the flash so the sketch's binary must be smaller than half of the flash. 
the size is checked against content length header of the http upload and htp error 413 "Payload Too Large" should be returned to upload tool.

burnt-past, 2021-06-02:
I have code the most similar to the "OTASketchDownload.ino" example, but I am not using the HTTP portion. I am using a USB disk that has the firmware on it. So I would never see that error.

Is there no way to remap the temp firmware location during a write, so that it is placed at a higher location?

burnt-past, 2021-06-02:
I was able to make this work by changing:
`maxSketchSize = (MAX_FLASH - bootloaderSize) / 2;`
to
`maxSketchSize = (MAX_FLASH - bootloaderSize) - 40000UL;`

This gives me plenty of space for what I am doing, though it is a corner case. I was trying to find a way to check for the last address used in application flash, but it seems I'm not finding what I'm looking for. This sloppy fix will do for now.

closed 2021-06-02 by burnt-past

jandrassy, 2021-06-03:
> I was trying to find a way to check for the last address used in application flash. 

`MAX_FLASH - bootloaderSize`

burnt-past, 2021-06-03:
I meant of the currently loaded application. I ended up just setting a hard value. This way it starts writing after the current firmware. Upon reboot it runs again but with smaller application usage and starts writing at a much lower point.