JAndrassy/ArduinoOTA #99: OTASketchDownload: How are you replacing the running code?

ThomHehl, 2021-07-13:
I'm trying to do something similar to OTASketchDownload only I'm downloading from bluetooth and writing to InternalStorage. It runs for a while and then crashes. I'm guessing where it overwrites the running code. How are you avoiding this issue?

jandrassy, 2021-07-13:
InternalStorage uses upper half of the flash to store the transported binary. on apply() the application binary is copied to required address. on ARM a library function in RAM copies the flash pages. on AVR a function in enahanced Optiboot copies the flash pages.
did you read the doc? https://github.com/jandrassy/ArduinoOTA#readme

ThomHehl, 2021-07-13:
> Links:

Thanks, that's a big help. How do I create the binary?