JAndrassy/ArduinoOTA #103: not working with Ethernet library

miksumin, 2021-07-29:
It seems it is not working properly with Ethernet library

miksumin, 2021-07-29:
I am using MegaCore with EthernetENC library on ATmega2561 with ENC28j60 chip

jandrassy, 2021-07-29:
https://github.com/jandrassy/ArduinoOTA#troubleshooting

miksumin, 2021-07-29:
Everything done as you wrote in OTA Upload from IDE without 'network port'
IDE output:
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
Flashing sketch ... Error flashing the sketch

jandrassy, 2021-07-29:
try this
Connect the board to computer and add in WiFiOTA.cpp to sendHttpResponse

  Serial.print("OTA upload response: ");
  Serial.println(status);

to see the response message.

miksumin, 2021-07-29:
OTA upload response: Unauthorized

miksumin, 2021-07-29:
Using as in example:
ArduinoOTA.begin(Ethernet.localIP(), "Arduino", "password", InternalStorage);

jandrassy, 2021-07-29:
did you copy extras/avr/platform.local.txt to MegaCore? the upload over 'programmer' has the password in upload command in platform.local.txt

miksumin, 2021-07-29:
Have copied the content of extras/avr/platform.local.txt to MegaCore platform.txt file. It contains the following:

# This configuration file supports the general ArduinoOTA library https://github.com/jandrassy/ArduinoOTA
## Create output (bin file)
recipe.objcopy.bin.pattern="{compiler.path}{compiler.elf2hex.cmd}" -O binary {compiler.elf2hex.extra_flags} "{build.path}/{build.project_name}.elf" "{build.path}/{build.project_name}.bin"
tools.avrdude.network_cmd={runtime.tools.arduinoOTA.path}/bin/arduinoOTA
tools.avrdude.upload.network_pattern="{network_cmd}" -address {serial.port} -port 65280 -username arduino -password "{network.password}" -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b
## arduinoOTA as programmer. add entries with {ip} into programmers.txt
tools.arduinoOTA.cmd={runtime.tools.arduinoOTA.path}/bin/arduinoOTA
tools.arduinoOTA.program.params.verbose=
tools.arduinoOTA.program.params.quiet=
tools.arduinoOTA.program.pattern="{cmd}" -address {ip} -port 65280 -username arduino -password password -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b

So where is the password?

miksumin, 2021-07-29:
Also tried to copy file platform.local.txt to MegaCore directory and got the following:
Connecting to board ...  done
Uploading sketch ...  done
An error occurred while uploading the sketch
Flashing sketch ... Error flashing the sketch

miksumin, 2021-07-30:
Please help

jandrassy, 2021-07-30:
still "OTA upload response: Unauthorized"?

set the IDE in Preferences to upload verbose mode and then post here the upload command

miksumin, 2021-07-30:
upload verbose mode in IDE is set and the output I posted above:
Connecting to board ... done
Uploading sketch ... done
An error occurred while uploading the sketch
Flashing sketch ... Error flashing the sketch

miksumin, 2021-07-30:
upload command:
...\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA -address 192.168.0.125 -port 65280 -username arduino -password password -sketch ...\AppData\Local\Temp\arduino_build_123594/MegaD-2561.ino.bin -upload /sketch -b

miksumin, 2021-07-30:
So what I'm doing wrong or what is going wrong in my case?

jandrassy, 2021-07-30:
still "OTA upload response: Unauthorized"?

miksumin, 2021-07-30:
Did you read my previous posts?

jandrassy, 2021-07-30:
> 
> 
> Did you read my previous posts?

yes. upload command is ok. does WiFIOTA always print Unauthorized?

miksumin, 2021-07-30:
WiFIOTA gives nothing output

jandrassy, 2021-07-30:
you use OTEthernet example without changes?

miksumin, 2021-07-30:
Yes, the code is identical OTEthernet example

jandrassy, 2021-07-30:
it is strange that it continues with "Flashing sketch" after "An error occurred while uploading the sketch".
add some additional output to the OTA uploaded sketch to see if it is applied.

miksumin, 2021-07-30:
Sketch is not starting after the above OTA upload error

jandrassy, 2021-07-30:
did you burn Optiboot from current version of MegaCore?

miksumin, 2021-07-30:
I have to use existing bootloader

jandrassy, 2021-07-30:
> I have to use existing bootloader

Optoboot with copy_flash_pages function is requirement.

miksumin, 2021-07-30:
How I can burn Optiboot?

jandrassy, 2021-07-30:
> How I can burn Optiboot?

with AVR programmer hardware or https://www.arduino.cc/en/Tutorial/BuiltInExamples/ArduinoISP

miksumin, 2021-08-19:
Tried to burn Optiboot using FT232R USB UART connected to ATmega2561 SPI without success:

         Programmer Type : ftdi_ft2xx
         Description     : Arduino: FT232R connected to ISP
         Pin assignment: 0..7 = DBUS0..7
           VCC     =  (not used)
           BUFF    =  (not used)
           RESET   =  7
           SCK     =  5
           MOSI    =  6
           MISO    =  3
           ERR LED =  (not used)
           RDY LED =  (not used)
           PGM LED =  (not used)
           VFY LED =  (not used)

avrdude: SPI bitclk 75000 -> FT baudrate 150000
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0xff00ff
avrdude: Expected signature for ATmega2561 is 1E 98 02
avrdude: erasing chip
avrdude: SPI bitclk 75000 -> FT baudrate 150000
avrdude: Program enable command not successful. Retrying.
avrdude: Program enable command not successful. Retrying.
avrdude: Program enable command not successful. Retrying.
avrdude: Program enable command not successful. Retrying.
avrdude: Device is not responding to program enable. Check connection.

Me be you can help me find a way to solve this?

jandrassy, 2021-08-19:
> Tried to burn Optiboot using FT232R USB UART connected to ATmega2561 SPI without success:

You can't burn the bootloader  with USB UART

closed 2021-08-19 by jandrassy

miksumin, 2021-08-22:
I had to burn Optiboot from current version of MegaCore with copy_flash_pages function enabled. Still getting an error uploading the sketch:
Connecting to board ...  done
Uploading sketch ...  done
An error occurred while uploading the sketch
Flashing sketch ... Error flashing the sketch

miksumin, 2021-08-23:
No any response from sendHttpResponse

miksumin, 2021-08-23:
Going restart after request received

miksumin, 2021-08-23:
Connecting to board ... done
Uploading sketch ... <restart> done
An error occurred while uploading the sketch
Flashing sketch ... Error flashing the sketch

miksumin, 2021-08-23:
The same result using your example "OTEthernet.ino"

jandrassy, 2021-08-24:
I will test it with MegaCore version of the bootloader. 
did you use the latest vesion of MegaCore? Optiboot in older versions doesn't have copy_flash_pages function

miksumin, 2021-08-24:
Yes, I used the latest one

jandrassy, 2021-08-29:
I tested it and everything works. But your uploads look like doubled. There shouldn't be "Flashing sketch" twice or "Flashing sketch" after failed upload.

miksumin, 2021-08-29:
I am using ATmega2561 with ENC28j60 ethernet burned with optiboot_flash bootloader and your example "OTEthernet.ino" compiled in Arduino 1.8.7 using last version of MegaCore with EthernetENC library. What I am doing wrong?

jandrassy, 2021-08-29:
do you have upload verbose mode in IDE Preferences checked?

miksumin, 2021-08-30:
Yes, of cause

jandrassy, 2021-11-18:
this could be related to enc28j60 and uIP stack. it can't finish sending the response to the upload tool and close the connection.
try to add a beforeApply function.
above setup define function
```
void beforeApplyUpdate() {
  for (int i = 0; i < 4; i++) {
    Ethernet.maintain();
    delay(UIP_PERIODIC_TIMER);
  }
}
```
in `setup()` add
```
ArduinoOTA.beforeApply(beforeApplyUpdate);
```