JAndrassy/ArduinoOTA #105: Still getting InternalStorage error

ActiveByte, 2021-08-01:
Hey there,

I flashed the bootloader to the Uno/Nano/Mini (Big Optiboot) board. (i have a Arduino Nano ATmega328P

Afther that i'm still getting the error:
'InternalStorage' was not declared in this scope

While using
ArduinoOTA.begin(Ethernet.localIP(), "arduino", "password", InternalStorage);

When i try to compile with the arduino mega optiboot board it does compile correctly. (but offcourse it can't upload that)
Is there maybe a problem with the optiboot file for Uno/Nano/mini?

Thanks

jandrassy, 2021-08-01:
because
https://github.com/jandrassy/ArduinoOTA/blob/bdfddb6f839e02e8eb062d9d2903bbd9e0adbf3c/src/ArduinoOTA.h#L23-L27

ActiveByte, 2021-08-01:
Thanks for the fast reply!

What does that mean, what can i do to use ArduinoOTA on my Nano with eth shield?

ActiveByte, 2021-08-01:
Oh now i see...

ActiveByte, 2021-08-01:
I did a bit of digging in the code, does it mean that my arduino does not have enough memory?

jandrassy, 2021-08-02:
you need half of the flash to store the binary. if you use some networking library to transport the binary, then the binary is larger then half of the flash of ATmega328. that is why only ATmega with at least 64kB flash is supported in the ArduinoOTA library.
if you have a way to transport the binary which doesn't make the sketch large, then you can use InternalStorageAVR directly

closed 2021-11-18 by jandrassy