JAndrassy/ArduinoOTA #114: Samd21 OTA over WiFi (ESP32)

Fahadk7, 2021-11-30:
Hi,
I have been working on a project which requires samd21 and esp32 together for an IoT device, connected over AWS
I am aware of how to perform OTA for esp32 with AWS libraries but need some guidance to do the same for SAMD21 connected over serial UART to esp32.
Any help is appreciated, as this is my first time working with SAMD chips
Will this library support the operation?
Do I need to download the firmware(in binary format) from AWS and store it in esp32 flash memory(16MB version) and then double reset the slave mcu SAMD and write the firmware over UART? Or what are the available altenatives?

Thanks in advance,
Fahad

jandrassy, 2021-11-30:
with ArduinoOTA the SAMD can do self-update, but only with an Arduino networking library (Enthernet, EthernetENC, WiFi101, WiFiNINA, WiFiEspAT). if you networking part is in the esp32 and the SAMD communicates with over Serial then you can't use ArduinoOTA for SAMD.

you could use the InternalStorage object of ArduinoOTA library to store and apply the update, but the transport of the binary is on you. see the OTASketchDownload example.

Fahadk7, 2021-12-18:
Thanks for the advice Juraj!

Yes, you are correct, the networking part is on esp32. I had thought that I would download the binary into esp32 and send the data over to SAMD either in an array or JSON over serial communication

Would this method work theoretically? (I understand the chance of data loss could I perform a checksum after receiving the data to verify?)

and then as you suggested, store it in InternalStorage and apply it?

I will try and work this with the reference example OTASketchDownload. Thanks a lot!

closed 2021-12-28 by jandrassy