JAndrassy/ArduinoOTA #116: SAMD boards package 1.8.12 breaks OTA?

stealth-ultd, 2021-12-02:
Just upgraded to SAMD boards package 1.8.12 – this seems to break OTA. (Also updated: Adafruit_BusIO, Adafruit_LED_Backpack_Library, Time)

>> Behaviour: After the pop-up for password input, the compiler enters a loop (back to pop-up), and displays the following errors:

java.io.IOException: Cannot run program "{tools.arduino_ota.cmd}": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.GenericNetworkUploader.uploadUsingPreferences(GenericNetworkUploader.java:99)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2055)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 10 more

>> Interestingly, rolling back to 1.8.11 does not resolve the issue, but gives a different set of errors:

fatal error: runtime: bsdthread_register error

runtime stack:
runtime.throw(0x13693d0, 0x21)
	/ssd/go/src/runtime/panic.go:616 +0x81 fp=0x7ff7bfeff958 sp=0x7ff7bfeff938 pc=0x1028be1
runtime.goenvs()
	/ssd/go/src/runtime/os_darwin.go:129 +0x83 fp=0x7ff7bfeff988 sp=0x7ff7bfeff958 pc=0x1026763
runtime.schedinit()
	/ssd/go/src/runtime/proc.go:501 +0xd6 fp=0x7ff7bfeff9f0 sp=0x7ff7bfeff988 pc=0x102b4d6
runtime.rt0_go(0x7ff7bfeffa20, 0xf, 0x7ff7bfeffa20, 0x1000000, 0xf, 0x7ff7bfeffbd8, 0x7ff7bfeffc34, 0x7ff7bfeffc3d, 0x7ff7bfeffc4a, 0x7ff7bfeffc50, ...)
	/ssd/go/src/runtime/asm_amd64.s:252 +0x1f4 fp=0x7ff7bfeff9f8 sp=0x7ff7bfeff9f0 pc=0x1050fe4

>> Removing and reinstalling ArduinoOTA does not resolve the issue.

Much appreciate the OTA feature, so hopefully, this can get fixed :-)

jandrassy, 2021-12-03:
I can't test it right now with 'network port', but I could test that the method with a fake programmer works. I recommend it.
https://github.com/jandrassy/ArduinoOTA#ota-upload-from-ide-without-network-port

note: if you use upload with programmer you can suppress the network port propagation with `#define NO_OTA_PORT` before `#include <ArduinoOTA.h>`

jandrassy, 2021-12-03:
is it this problem https://github.com/arduino/ArduinoCore-samd/pull/656 ?

stealth-ultd, 2021-12-04:
Thanks for looking into this. I can test the 'network port' method after the weekend (right now I am out of town). Will be back on this.

stealth-ultd, 2021-12-06:
From what I understand the "Fix legacy upload.network_pattern rules" is not the same issue. It is related to the update to Arduino Java IDE 1.8.16, while I am using Arduino IDE 1.8.13 with SAMD 1.8.12.

Will now try the fake programmer method.

jandrassy, 2021-12-06:
> From what I understand the "Fix legacy upload.network_pattern rules" is not the same issue. It is related to the update to Arduino Java IDE 1.8.16, while I am using Arduino IDE 1.8.13 with SAMD 1.8.12.

it is an issue of 1.8.12 SAMD which was released 20 days ago.

stealth-ultd, 2021-12-06:
For this to be properly resolved ("out of the box"), do we have to wait for an update to the SAMD package?

(I see that in 1.8.12 SAMD platforms.txt the "# Arduino OTA" section is already included, but different from the code suggested in ArduinoOTA/extras/samd/platform.local.txt)

jandrassy, 2021-12-06:
> For this to be properly resolved ("out of the box"), do we have to wait for an update to the SAMD package?
yes

> (I see that in 1.8.12 SAMD platforms.txt the "# Arduino OTA" section is already included, but different from the code suggested in ArduinoOTA/extras/samd/platform.local.txt)

the SAMD package has support for 'network port'  OTA so extras/platform.local.txt only adds support for the `fake programmer`

therwastl07, 2022-04-16:
I have the same issue. IDE of 1.8.13, boards as above 1.8.12 as the latest one
When I go "fake programmer" I get the message: missing configuration parameter "program.params.verbose" and the upload stops

closed 2022-04-16 by jandrassy