JAndrassy/ArduinoOTA #123: Arduino Nano RP2040 Connect

rocrail, 2022-01-02:
Hi,
trying to use the example WiFi101_OTA brings a link error:

```
WiFi101_OTA.ino:57: undefined reference to `InternalStorage'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Nano RP2040 Connect.
```
How to solve this?

jandrassy, 2022-01-02:
the RP2040 support is not in a released version yet. you have to download a zip from GitHub (use the green "Code" button)

rocrail, 2022-01-02:
Thanks,
but this brings another error:
```
Documents/Arduino/libraries/ArduinoOTA/src/utility/rp2_flash_boot.c:25:10: fatal error: RP2040.h: No such file or directory
 #include <RP2040.h> // CMSIS
          ^~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Nano RP2040 Connect.

```

jandrassy, 2022-01-02:
do you use some older version of the Pico core?

rocrail, 2022-01-02:
No, the target is an Arduino Nano RP2040 Connect.
So I did not install the Pico Core SDK.

jandrassy, 2022-01-02:
ArduinoOTA for RP2040 doesn't work with the Arduino mbed core.

from README
>RP2040 boards with [Pico core](https://github.com/earlephilhower/arduino-pico)

rocrail, 2022-01-02:
OK.
But the Raspberry Pi Pico has no WiFi...

jandrassy, 2022-01-02:
> OK. But the Raspberry Pi Pico has no WiFi...

how is that a problem?

rocrail, 2022-01-02:
Is not a problem.
My issue title is "Arduino Nano RP2040 Connect"
The Arduino Nano RP2040 Connect has the same CPU as the Pico.
The Pico I have, does not feature any WiFi, so it makes no sense for me to support Pico "Over The Air".
Or did I miss something?

jandrassy, 2022-01-02:
[image]

jandrassy, 2022-01-02:
nor does MKR Zero have WiFi
[image]

closed 2022-02-09 by jandrassy

jmdodd95682, 2022-07-23:
Do you know if the Nano RP2040 Connect (with the mbed core) will ever be supported by ArduinoOTA?  Just curious.  I'm ripping out the code for supporting OTA and I want to know if its worth preserving the code or not.

Thanks!

JAndrassy, 2022-07-23:
RP2040 is supported with the earlephilhower/arduino-pico core with ArduinoOTA library.

With mbed core Nano RP2040 Connect can be OTA updated from IoT Cloud. It stores the binary to file system on the flash and then after reset it is read and applied with the SFU library. You can use the SFU library the same way. I don't have Nano RP2040 Connect so I can't develop OTA upload to mbed file system.

AcThPaU, 2023-01-26:
I don't think the mDNS works. Is it possible to upload using fake programmer method? The Arduino-pico core has very different menu.

lcoffin, 2023-02-25:
Sorry if this feels like being pestered, but any plans on supporting the stock Arduino Nano RP2040 Connect? Without having to mess around with changing the core? I'm happy to support the acquisition of a board if that would help! :)

JAndrassy, 2023-02-26:
using arlephilhower/arduino-pico core is not messing up. it is the primary Arduino core for the RP2040.
the Arduino mbed core for Nano Connect only supports Nano Connect

lcoffin, 2023-02-26:
Sorry, I didn't mean to imply that switching to the Philhower Core was "messing up". Just that the out of the box core that comes with the Nano RP2040 Connect is the mbed one, so having to change the core is an extra step. For microcontroller newbies like myself, with my code written using the mbed core, I'm not particularly eager to go through the steps of switching cores, possibly breaking my existing code. I'd much more prefer finding an OTA solution that works with what I have -- the default core that comes with the board. (And that doesn't require the Arduino IOT cloud.) So far, your OTA library appears to be the closest to a working solution for that, it just doesn't support the RP2040 Connect flash storage -- if I understand the issue correctly.

JAndrassy, 2023-09-20:
the reasons are technical, not political

Alain-Godo, 2023-09-20:
Is difficult? I'm in the best disposition to help if needed. 👍

JAndrassy, 2023-09-21:
not difficult. the flash writing API is not available and the RTOS may conflict with flash writing in ArduinoOTA style

JAndrassy, 2023-09-21:
you can use the SFU library with Nano RP2040. it is not documented, but int ArduinoIoTCloud library you can see how they store the binary for the SFU to load at boot.
https://github.com/arduino-libraries/ArduinoIoTCloud/blob/master/src/utility/ota/OTA-nano-rp2040.cpp

JAndrassy, 2023-09-22:
I forgot. It is already solved
https://github.com/JAndrassy/ArduinoOTA/blob/master/examples/Advanced/CustomStorageNanoConnect/CustomStorageNanoConnect.ino

lcoffin, 2023-09-24:
@Alain-Godo Let me know if you run into any issues. I have it working on a few RP2040 Connects. The WiFi on the board seems to be a bit wonky at times (esp on reconnects) so I've found the OTA can be tricky -- best bet is to try to do the OTA soon after starting the board. You might need to upgrade the arduinoOTA executable over what gets installed usually (though that may have changed since I did it several months ago) and increase the upload timeout. And use the Programmer to do the upload rather than the standard way.

BrettLHolmes, 2023-10-04:
@lcoffin great to hear, did you have some working sample code, and anything special to get it working. 
can you explain what you mean by upgrade the arduinoOTA executable over what gets installed usually.

Alain-Godo, 2023-10-04:
@BrettLHolmes Using the Arduino IoT Cloud code philosophy is possible to make the OTA update. I finally did it. I can't share code due to professional issues but I can tell you how if u want. Let me know.

JAndrassy, 2023-10-04:
@BrettLHolmes https://github.com/JAndrassy/ArduinoOTA/blob/master/examples/Advanced/CustomStorageNanoConnect/CustomStorageNanoConnect.ino

BrettLHolmes, 2023-11-05:
Thankyou @JAndrassy  and @lcoffin for you time to respond - i will be attempting this soon because I have a few in production use

BrettLHolmes, 2023-11-06:
Hi both
A little feedback from me

i have all the configuration done, the ide compiles and starts the upload, it says it completes but the it says failed to flash and exits

i have tried to run from the command line and even extended the timeouts

in looking at a packet trace (for which i have no basis for a good trace) there are a lot of retransmits right at the end befor the flashing fails

it could be something to do with the size of the binary file which i do not know what the max can be or how to calculate.  I plan on trying to upload a small sketch because the one i am currently trying to upload may be a little bloated tft display (with some icons)and a 433 mhz radio driver along with mqtt but this may take me a while to try because i lack the time

lcoffin, 2023-11-06:
@BrettLHolmes Have you tried the CustomStorageNanoConnect example sketch referenced above?

BrettLHolmes, 2023-11-24:
@lcoffin , i apologies i did not look at your question.

it said was i using the 3xample but i was upkntegrating it into my own sketch which could have cause other issues.

i have since used the example verbatim and still not succeeded.

two issues
I do not get the programmer menu to be able to push to the board from the menus 
I have however pushed the image to the board via the command line which still fails  after xonnexting and uploading but it fails toward the end

in the last part there is no clear indication why it failed

BrettLHolmes, 2023-12-03:
Hello both

I wanted to let you know that I had success with using the OTA updates Library on the RP2040 Nano
there were 2  issues: (and for anyone else trying)

1)
the IDE gives an Error with the "programmers.txt" definitions as they are above:
i had to change the line :
arduinoOTA245.program.**default**=arduinoOTA ,-- This line did not work gives error ("Upload error: Property 'program.tool.default' is undefined") when trying to upload
arduinoOTA245.program**_.tool.default_**=arduinoOTA <--Changing to this seems to work

2)
I also had to change to running a nightly build of version 2.2.2 of Arduino IDE 2.2 for the mac which also seems to fix the error with the programmers items not appearing from the Tools menu. - A note the above error was even on the released version of the IDE Version 2.2.1
 
Thank you both for your help and guindance - much appreciated.

JAndrassy, 2023-12-03:
> arduinoOTA245.program.tool.default=arduinoOTA 

README has it since August