JAndrassy/ArduinoOTA #124: Error flashing when using ESP32-S2-WROVER (Saola-1R module)

davorvr, 2022-01-04:
Hello, I keep getting an error when trying to flash via WiFi. I'm using a modified WiFi101_OTA example which connects successfully:
```
/*

 This example connects to an WPA encrypted WiFi network.
 Then it prints the  MAC address of the Wifi shield,
 the IP address obtained, and other network details.
 It then polls for sketch updates over WiFi, sketches
 can be updated by selecting a network port from within
 the Arduino IDE: Tools -> Port -> Network Ports ...

 Circuit:
 * WiFi shield attached

 created 13 July 2010
 by dlf (Metodo2 srl)
 modified 31 May 2012
 by Tom Igoe
 modified 16 January 2017
 by Sandeep Mistry 
 */

#include <SPI.h>
#include <WiFi.h>
#include <ArduinoOTA.h>

#include "config.h"
///////please enter your sensitive data in the Secret tab/arduino_secrets.h
/////// Wifi Settings ///////
char ssid[] = WIFI_SSID;      // your network SSID (name)
char pass[] = WIFI_PASS;   // your network password

//int status = WL_IDLE_STATUS;

void setup() {
  //Initialize serial:
  Serial.begin(115200);
  //Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  if (!WiFi.config(LOCAL_IP, GATEWAY, SUBNET, DNS_PRIM, DNS_SECOND)) {
    //Serial.println("STA Failed to configure");
  }

  // attempt to connect to Wifi network:
  WiFi.begin(ssid, pass);
  for (int i=0; i<600; i++) {
    if (WiFi.status() == WL_CONNECTED) break;
    delay(100);
  }
  if (WiFi.status() != WL_CONNECTED) {
    //Serial.println("Failed to connect. Restarting...");
    ESP.restart();
  } else {
    //Serial.println("Connected.");
  }

  // start the WiFi OTA library with internal (flash) based storage
  ArduinoOTA.begin(WiFi.localIP(), "Arduino", "password", InternalStorage);

  // you're connected now, so print out the status:
  printWifiStatus();
}

void loop() {
  // check for WiFi OTA updates
  ArduinoOTA.poll();

  // add your normal loop code below ...
  delay(1000);
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}

```
The error is simply as follows:
```
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
```
The module only has 4 MB of flash, but the sketch is only about 600 kB so I'm not sure if this would be the issue. I have deleted the arduino-esp32 ArduinoOTA library directory before using this library.

jandrassy, 2022-01-04:
sorry but why do you use this library?

please see https://github.com/jandrassy/ArduinoOTA/issues/56 for esp32 problems

davorvr, 2022-01-04:
The WiFi library? No idea, I was under the impression that in this example I am using the WiFi library from Espressif's esp32 package. Is it better to use the vanilla Arduino WiFi library? Thank you for the quick reply and the related issue, I will take a look.

jandrassy, 2022-01-04:
I meant the esp32 native WiFi library. The only one which would work with an esp32.
why do you want to do OTA with my library if you use the esp32 WiFi?
to use this library on esp32 makes sense only if you want to do OTA over the Arduino Ethernet library on esp32

davorvr, 2022-01-04:
Ah, my interest in this library was due to the possibility of uploading to a device by specifying its IP instead of having to rely on mDNS.

jandrassy, 2022-01-04:
so yes. the espota can not be configured as fake programmer because it is not installed as a separate tool. but you could run it from command line.

jandrassy, 2022-01-05:
so it is possible
https://github.com/esp8266/Arduino/pull/8432/files

davorvr, 2022-01-05:
So I could use your arduinoOTA tool to upload a binary from the command line to a board that uses the default ESP32 ArduinoOTA library?

> so it is possible https://github.com/esp8266/Arduino/pull/8432/files

This is great, thank you!!

jandrassy, 2022-01-05:
> So I could use your arduinoOTA tool to upload a binary from the command line to a board that uses the default ESP32 ArduinoOTA library?
> 
> > so it is possible https://github.com/esp8266/Arduino/pull/8432/files
> 
> This is great, thank you!!

it is espota, not arduinoOTA, but yes, it is the fake programmer method

closed 2022-01-05 by jandrassy

davorvr, 2022-01-05:
Excellent! Thank you very much and sorry for bothering with something that is basically off-topic and not an issue with your library at all.

jandrassy, 2022-01-05:
I must thank you. I wanted to do the esp8266 fake programmer a long time ago, but I thought it is not possible because the espota is not installed by the IDE, it is only a file in the core's folder. But yesterday solving timeout problems of arduinoOTA I discovered that I can refer any program in the fake programmer definition. But I didn't think about esp8266 then. Only today as I saw this page again I had the idea.

davorvr, 2022-01-05:
Hehe, that's wonderful, I'm really glad! Thanks again!