JAndrassy/ArduinoOTA #129: Error when 'Upload using programmer' to Mega 2560 with optiboot bootloader

RStaes, 2022-01-11:
Hello,

i have this problem: I have an Arduino Mega 2560 and flashed the optiboot bootloader to it with an uno as programmer. All went fine. I uploaded my sketch, see code below, with use of the serial port. When upload is ready i can ping the Mega. Then, after a restart of Arduino IDE, i can't see the network port. But when i choose for programmer 'Arduino OTA 192.168.1.119' and click 'Upload using programmer' I get this error:

Can you help me out? 

```
Arduino: 1.8.13 (Windows 10), Board: "Optiboot on Mega2560"

Sketch uses 41456 bytes (15%) of program storage space. Maximum is 261120 bytes.

Global variables use 3174 bytes (38%) of dynamic memory, leaving 5018 bytes for local variables. Maximum is 8192 bytes.

C:\Users\Robin\AppData\Local\Arduino15\packages\Optiboot\hardware\avr/tools/arduinoOTA.exe -address 192.168.1.119 -port 65280 -t 30 -username arduino -password password -sketch C:\Users\Robin\AppData\Local\Temp\arduino_build_384939/Zoneverwarming3.ino.bin -upload /sketch -b 

Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
Error flashing the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Sketch:
```
#include <Ethernet.h>
#include <ArduinoOTA.h>
#include <SPI.h>
#include "PubSubClient.h"
#include <OneWire.h>
#include <DallasTemperature.h>
#include <avr/wdt.h> // for Watchdog

#define ONE_WIRE_BUS 2
#define CLIENT_ID       "meterkast"
#define PUBLISH_DELAY   10000
#define FAST_MODE // 50 times faster execution; consider to disconnect your real CV/Pump!

// In Normal operation to loop runs 10 times per second; so 10 counts/second (600 represents ca 1 minute)
#define VALVE_TIME             3000L    // 5 minutes to open/close a valve (on the safe site; takes typically 3 to 5 minutes)
#ifdef  FAST_MODE
#define PUMP_MAINTENANCE_TIME  108000L // For evaluation, activates Floor Unit pump maintenance run once per 4 minutes (time stamp 3 hours)
#else
#define PUMP_MAINTENANCE_TIME  1300000L // Activates Floor Unit pump maintenance run once per 36 hours. Needed to keep pump working
#endif
#define PUMP_ACTIVATION_TIME   5000L    // Activates the pump for ca 8 minutes (10 seconds in test mode)
#define COOLDOWN_TIME          18000L   // When heating is done, continue water circulation for another 30 minutes (40 seconds in test mode)
                                        // This enables further dissipation the heat into the floor (typically takes 15 to 30 minutes)

#include "./Devices.h" // valves, pumps, thermostat classes (use the constants defines above)

#define HEATER_PIN      33 // output to a Relay that is wired with the Thermostat input of your heating system
#define FU_PUMP_PIN     25 // output to a Relay that switches the Floor Unit Pump
#define PORCH_RELAY     24 // Zone 1: output to a Relay that controls the Valve(s)
#define LIVING_RELAY    27 // Zone 2: output to a Relay that controls the Valve(s)
#define PLAYROOM_RELAY  26 // Zone 3: output to a Relay that controls the Valve(s)
#define OFFICE_RELAY    29 // Zone 4: output to a Relay that controls the Valve(s)
#define BATHROOM_RELAY  32 // Zone 5: output to a Relay that controls the Valve(s)
#define STORAGE_RELAY   31 // Zone 6: output to a Relay that controls the Valve(s)
#define UPSTAIRS_RELAY  34 // Zone 7: output to a Relay that controls the Valve(s)

#define HEATING_LED     41 // On when heating, Alternates during cooldown, is Off in idle mode
#define INDICATION_LED  13 // Alternates the on board LED to indicate board runs; can be easily removed to free an extra IO pin!!

// Configure the Floor Unit Zones/rooms. Each zone/room owns a name, valve and thermostat:
#define NR_ZONES 7
#define NR_ZONETHERMOSTATS 7

[... 605 lines cut ...]
      }
      else{
        Zones[i].Action = "idle";      
      }
    }
    else if (Zones[i].ModeState == "off"){
        Zones[i].thermostat = false;
        Zones[i].Action = "off";
    }
  }
  PrintZones();
  sendData();
}
void PrintZones(){
  for (int i=0; i<NR_ZONES; i++) {
    Serial.println(Zones[i].name);
    Serial.println(Zones[i].IsTemp);
    Serial.println(Zones[i].SetTemp);
    Serial.println(Zones[i].Action);
    Serial.println(Zones[i].Mode);
    Serial.println(Zones[i].ModeState);
    Serial.println("");
  }
}
```

jandrassy, 2022-01-11:
I guess it is a timeout in the upload tool. 
my_boards now has a version of the upload tool with timeout parameter and in avr platform.txt 30 seconds is set as timeout.
get the files.
If you don't want to use an exe downloaded from internet, you can download[ the source code of arduinoOTA ](https://github.com/arduino/arduinoOTA/blob/master/main.go)and build it with the Go language builder (which is simple to install)

RStaes, 2022-01-11:
Hey thanks! Where do i put this files? Now i have them in 

`C:\Users\User\AppData\Local\Arduino15\packages\arduino\hardware\avr\tools.`

The platform.txt contains this line: 

`C:\Users\User\AppData\Local\Arduino15\packages\Optiboot\hardware\avr/tools/arduinoOTA.exe -address 192.168.1.119 -port 65280 -t 30 -username arduino -password password -sketch C:\Users\User\AppData\Local\Temp\arduino_build_384939/Zoneverwarming3.ino.bin -upload /sketch -b `

Note the / and the \ in the path. Is that right? I put this in the bottom of the platform.txt file with -t 30:

```
## Create output (bin file)
recipe.objcopy.bin.pattern="{compiler.path}{compiler.elf2hex.cmd}" -O binary {compiler.elf2hex.extra_flags} "{build.path}/{build.project_name}.elf" "{build.path}/{build.project_name}.bin"

build.preferred_out_format=bin

## Save hex
recipe.output.tmp_file={build.project_name}.{build.preferred_out_format}
recipe.output.save_file={build.project_name}.{build.variant}.{build.preferred_out_format}

tools.avrdude.upload.network_pattern="{network_cmd}" -address {serial.port} -port 65280 -username arduino -password password -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b

tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.with_bootloader.hex:i"

tools.arduinoOTA.cmd={runtime.hardware.path}/tools/arduinoOTA
tools.arduinoOTA.cmd.windows={runtime.hardware.path}/tools/arduinoOTA.exe
tools.arduinoOTA.program.params.verbose=
tools.arduinoOTA.program.params.quiet=
tools.arduinoOTA.program.pattern="{cmd}" -address {ip} -port 65280 -t 30 -username arduino -password password -sketch "{build.path}/{build.project_name}.bin" -upload /sketch -b
```

The platform.txt file is located here:

`C:\Users\User\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4`

jandrassy, 2022-01-11:
sorry, I modified only the 'programer' definition, not the standard network_pattern.

in the past it was not possible to setup the network_command with keys other than the "runtime.tools."
so I recommend you to use[ the fake programmer method](https://github.com/jandrassy/ArduinoOTA#ota-upload-from-ide-without-network-port)

RStaes, 2022-01-11:
Still not working. I have the same error as before: I even tried with a 60 seconds en 180 seconds timeout.

```
Arduino: 1.8.13 (Windows 10), Board: "Arduino Mega 2560 (Optiboot)"

Sketch uses 41442 bytes (15%) of program storage space. Maximum is 261120 bytes.

Global variables use 3172 bytes (38%) of dynamic memory, leaving 5020 bytes for local variables. Maximum is 8192 bytes.

C:\Users\User\Documents\Arduino\hardware\my_boards-master/tools/arduinoOTA.exe -address 192.168.1.119 -port 65280 -t 60 -username arduino -password password -sketch C:\Users\User\AppData\Local\Temp\arduino_build_739379/Zoneverwarming3.ino.bin -upload /sketch -b 

Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
Error flashing the sketch
```
I'm missing something, but what?

jandrassy, 2022-01-11:
turn on upload verbose mode in IDE Preferences

jandrassy, 2022-01-11:
turn off the watchdog

RStaes, 2022-01-11:
Hero!! It was the watchdog!

RStaes, 2022-01-11:
And with ArduinoOTA.beforeApply I can disable the watchdog and still use it in my sketch. Thank you very much!!

jandrassy, 2022-01-11:
> And with ArduinoOTA.beforeApply I can disable the watchdog and still use it in my sketch. Thank you very much!!

with a small sketch a longer watchdog time may work. I use watchdog interrupt in my Regulator project to extended the watchdog time. I activate a reset in second interrupt so I have a 16 seconds watchdog

closed 2022-01-11 by jandrassy