JAndrassy/ArduinoOTA #133: Sketch flashing is done but Arduino runs the old sketch.

RStaes, 2022-02-16:
Hello,

i have this problem: I have an Arduino Mega 2560 and flashed the optiboot bootloader to it with an Uno as programmer. All went fine. When i use 'Upload using programmer' I get no error but the message below telling me all went fine. But when the Mega restart it just runs the old sketch.

```
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
OK
Sketch uploaded successfully
```
What am I doing wrong?

Sketch:

```
#include <Ethernet.h>
#include <ArduinoOTA.h>
#include <SPI.h>
#include "PubSubClient.h"
#include <DallasTemperature.h>
#include <avr/wdt.h> // for Watchdog

#define CLIENT_ID       "meterkast"
#define PUBLISH_DELAY   10000
#define FAST_MODE // 50 times faster execution; consider to disconnect your real CV/Pump!

// In Normal operation to loop runs 10 times per second; so 10 counts/second (600 represents ca 1 minute)
#define VALVE_TIME             3000L    // 5 minutes to open/close a valve (on the safe site; takes typically 3 to 5 minutes)
#ifdef  FAST_MODE
#define PUMP_MAINTENANCE_TIME  108000L // For evaluation, activates Floor Unit pump maintenance run once per 4 minutes (time stamp 3 hours)
#else
#define PUMP_MAINTENANCE_TIME  1300000L // Activates Floor Unit pump maintenance run once per 36 hours. Needed to keep pump working
#endif
#define PUMP_ACTIVATION_TIME   5000L    // Activates the pump for ca 8 minutes (10 seconds in test mode)
#define COOLDOWN_TIME          18000L   // When heating is done, continue water circulation for another 30 minutes (40 seconds in test mode)
                                        // This enables further dissipation the heat into the floor (typically takes 15 to 30 minutes)

#include "./Devices.h" // valves, pumps, thermostat classes (use the constants defines above)

#define HEATER_PIN      33 // output to a Relay that is wired with the Thermostat input of your heating system
#define FU_PUMP_PIN     25 // output to a Relay that switches the Floor Unit Pump
#define PORCH_RELAY     24 // Zone 1: output to a Relay that controls the Valve(s)
#define LIVING_RELAY    27 // Zone 2: output to a Relay that controls the Valve(s)
#define PLAYROOM_RELAY  26 // Zone 3: output to a Relay that controls the Valve(s)
#define OFFICE_RELAY    29 // Zone 4: output to a Relay that controls the Valve(s)
#define BATHROOM_RELAY  32 // Zone 5: output to a Relay that controls the Valve(s)
#define STORAGE_RELAY   31 // Zone 6: output to a Relay that controls the Valve(s)
#define UPSTAIRS_RELAY  34 // Zone 7: output to a Relay that controls the Valve(s)

#define HEATING_LED     41 // On when heating, Alternates during cooldown, is Off in idle mode
#define INDICATION_LED  13 // Alternates the on board LED to indicate board runs; can be easily removed to free an extra IO pin!!

// Configure the Floor Unit Zones/rooms. Each zone/room owns a name, valve and thermostat:
#define NR_ZONES 7
#define NR_ZONETHERMOSTATS 7

byte mac[]    = {  0xDE, 0xED, 0xBA, 0xFE, 0xFE, 0xED };
IPAddress ip2(192, 168, 1, 64);
EthernetClient ethClient;
PubSubClient mqttClient(ethClient);
long previousMillis;
const int controlPin[] = {25,24,27,26,29,32,31,34,33,36,35,38,37,40,39,41}; // define pins
const int triggerType = LOW;// your relay type
int deviceCount = 0;
int Counter = 0;
bool Thermo           = false;  
float IsTemp          ;
float SetTemp         ;
String IsTopic        = "/is";
[... 509 lines cut ...]
      }
    }
    else if (Zones[i].ModeState == "off"){
        Zones[i].thermostat = false;
        Zones[i].Action = "off";
    }
  }
//  PrintZones();
  sendData();
}
void PrintZones(){
  for (int i=0; i<NR_ZONES; i++) {
    Serial.println(Zones[i].name);
    Serial.println(Zones[i].IsTemp);
    Serial.println(Zones[i].SetTemp);
    Serial.println(Zones[i].Action);
    Serial.println(Zones[i].Mode);
    Serial.println(Zones[i].ModeState);
    Serial.println("");
  }
}
void shutdownwdt() {
wdt_disable();
}
```

RStaes, 2022-02-16:
I’m using The latent from your “my hardware” folder. 
`https://github.com/jandrassy/my_boards/blob/master/avr/bootloaders/optiboot8/optiboot_atmega2560.hex`
I use Arduino IDE on Windows 10.

jandrassy, 2022-02-17:
turn off the watchdog

RStaes, 2022-02-17:
I did, same result.

jandrassy, 2022-02-17:
I had a problem with OTA caused by Linux shell some weeks ago, so I tested the OTA upload to Mega and after I solved the command line problem the upload worked without problems. Others sometime report problems and we solve them and it works. Usual problems are those mentioned in troubleshooting section, watchdog and self reset problems on some derivative Mega.

RStaes, 2022-02-17:
Ok, but it worked before... 

The error is not in the sketch I think. When i upload this sketch with USB it runs. Then when I change text in serialprint and upload the sketch with programmer it is still not working. I think something is wrong with the programmer.

jandrassy, 2022-02-17:
can you try the upload over network port?

RStaes, 2022-02-17:
How can I do that?

jandrassy, 2022-02-17:
with Ethernet library it should be possible to get the network port. try every trick you find with google.
I usually try to restart Arduino IDE, restart the network interface, restart the OS (computer). 
But maybe you need to install Bonjour into Windows.

RStaes, 2022-02-18:
I'm sorry to bother you again... 

Ok, now when I upload the sketch to the network port after compiling a window pops up where I have to enter a password. I guess it is "password" since that's the one in the sketch. Now I get this error: 

```
{network_cmd} -address 192.168.1.120 -port 65280 -username arduino -password password -sketch C:\Users\Robin\AppData\Local\Temp\arduino_build_844495/Zoneverwarming_boven.ino.bin -upload /sketch -b 
java.io.IOException: Cannot run program "{network_cmd}": CreateProcess error=2, Het systeem kan het opgegeven bestand niet vinden
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.GenericNetworkUploader.uploadUsingPreferences(GenericNetworkUploader.java:99)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2055)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=2, Het systeem kan het opgegeven bestand niet vinden
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more

```

On google I can not find something that causes this error.

jandrassy, 2022-02-18:
there is a bug in Arduino AVR core 1.8.4 with OTA settings. as you see {network_cmd} is not replaced. use 1.8.3 or do this fix:
https://github.com/arduino/ArduinoCore-avr/commit/525be9e725744c29f8c3b3fa4c216ce4f768e035

and AVR core 1.8.5 is already on its way to Boards Manager. maybe in a few days, maybe in a few hours.

closed 2022-04-24 by JAndrassy