JAndrassy/ArduinoOTA #134: WiFi OTA upload to Mega works unreliably

aryadas98, 2022-03-04:
I have been trying to use this library for OTA upload to Arduino Mega using an attached ESP8266. For the ESP I am using a NodeMCU board. My hardware connection is as follows:

| Arduino | NodeMCU |
| --- | --- |
| 5V | 5V |
| GND | GND |
| TX1 | RX |
| RX1 | TX |

I have flashed the Optiboot bootloader from the my_boards package onto the Mega. After this, USB upload with Optiboot works. I have also flashed AT 2 firmware onto the NodeMCU. After this, the WiFiESPAT sketches work. I can connect to the internet and send and receive data. So next, I flashed the example SerialWiFiOTA sketch to the Mega using USB. A network port showed up. I tried flashing the same sketch again, but this time using OTA. Sometimes it works, sometimes it doesn't. The error is usually:

```
Connecting to board ... failed!
Error flashing the sketch
```

Sometimes it is:
```
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
```

aryadas98, 2022-03-04:
On a side note, my project mainly requires network comms and preferably OTA update. I am willing to switch to some other board or architecture, if it would be easier. So any suggestions are welcome.

jandrassy, 2022-03-04:
AT2 has a very bad performance and is unable to make larger transports in or out.

aryadas98, 2022-03-05:
Thank you, it fixed the problem. OTA upload now works reliably.

closed 2022-03-05 by aryadas98