JAndrassy/ArduinoOTA #144: OTASketchDownload partitions

Bluebird-Software, 2022-03-18:
I use the OTASketchDownload.ino example to update via a HTTP server. The file is being downloaded successfully and the update works. 

I have the next partitions:

- nvs
- otadata
- app0
- app1
- spiffs

Are these updates being downloaded to the app0 / app1 partition, or the same partition as the main program?

jandrassy, 2022-03-18:
the esp32 Update library is used to store and apply the update binary

Bluebird-Software, 2022-03-21:
So how would you suggest the partitions. Do i need otadata and app0 and app1? or just 1 big app0 partition?

jandrassy, 2022-03-21:
see the esp32 Update library

closed 2022-03-21 by jandrassy

Bluebird-Software, 2022-03-21:
A little help would be appreciated. 
You mean this? https://github.com/espressif/arduino-esp32/tree/master/libraries/Update

How do i get the answer to my question?

jandrassy, 2022-03-21:
I don't know anything about esp32 partitions