JAndrassy/ArduinoOTA #145: Nano IOT 33 - WifiOTA Sketch Download

JEG8, 2022-03-21:
Hello,

I am using OTA sketch download using the example code provided (OTASketchDownloadWifi). 

The example sketch uses 38724 bytes (14%) of program storage space. I am trying to download a sketch that uses 148164 bytes (56%) of program storage space. Maximum storage on my device is 262144 bytes. I am receiving the following error.

**Update status code: 200
Server returned update file of size 147676 bytes
There is not enough space to store the update. Can't continue with update.**

Is there any way that I can make this work besides reducing the size of my update? 

Thanks for any help you can provide and thank you for this library.

jandrassy, 2022-03-22:
you can use SD card to store the update binary (or to store the data which are now in the sketch and make it big)

closed 2022-03-22 by jandrassy

JEG8, 2022-03-22:
Thank you. I had the same thought but when I try to use OTASketchDownload_SD for the Arduino Nano 33 IOT, it says the board is unsupported. I think this is due to the SDU library not supporting the board.

**\Arduino15\packages\arduino\hardware\samd\1.8.13\libraries\SDU\src\SDU.cpp:44:4: error: #error "Unsupported board!"
   #error "Unsupported board!"
    ^~~~~**

What do you recommend I use for storing the binary to the SD?

jandrassy, 2022-03-22:
https://github.com/arduino/ArduinoCore-samd/pull/645